Stephane-D/SGDK #31: Big sprites do not display correctly

gilvanmaia, 2015-06-10:
Hi there.

I'm trying to insert a sprite that is 13x13 tiles big. Data seems to be OK, but as soon as an animation is changed, the sprite partially blinks and also displays some bad tiles.

Is there something important that I am missing or is it really a bug?
By the way, I made some slides about basic function usage etc and could send these to you.

PS: did not find your e-mail therefore I am sending this message.

Stephane-D, 2015-06-11:
A 13x13 tiles sprite should fit in the sprite size limit of SGDK (which is a maximum of 20 hardware sprites and here you need 16). It may be related to the sprite cache size but if the first sprite frame is ok then it should be able to display other frame as well.
Can you provide me s small test sample so i can reproduce the problem ?

Also i would be really glad to put your slides as part of the SGDK tutorial ! As you may have noticed current tutorials are outdated and minimalists :-/

gilvanmaia, 2015-06-11:
Wow, that reply was really fast. =)
Sorry about my timing, but we got a blackout yesterday and the internet got really slow.

Here goes more detail:
 - every frame in my spritesheet has 13x13 tiles, and a total of 8 animations. These are for initial tests  aiming to build a fighting game or a "beat them up";
 - the longer animation is a walking sequence with 8 frames;
 - confirmed that problems occur when the animation changes.
 
First I tried to insert the sprite directly in my code. When the error occured, I assumed it was a problem with the code I wrote, and decided to test things in the sprite sample.
Here is the result:
[image]

I thought it may be the cache size. I tried to increase it to 512 and the problem was still in there.
Just let me get permission from the artist to send you the original image.

Best regards, and I will come back soon.

gilvanmaia, 2015-06-11:
Hello again,  

=D

This is the original spritesheet.
[image]

And this is the rescaled spritesheet (no care at all with art adjustments here, of course):

[image]

Let me know whether there is problems and how to fix them.
I will do my best to explain how users can avoid such problem.

Thank you in advance,

Stephane-D, 2015-06-14:
Thanks for the sprite sheet :) I just tested in the sprite example and i can reproduce the problem if i keep sprite size cache to 256 but the problem disappears if i use `SPR_init((13*13*2) + 100)` for instance. But i modified the sprite code a bit in my WIP, that may explain the difference. Anyway even at 256 we should have enough tiles and i noticed some others update bug so i will investigate that :)
Thanks for reporting ! By the way in the meantime, try also to disable the compression for the sprite as it consume a lot of cpu time ;)

gilvanmaia, 2015-06-16:
Thanks for all those tips. Works just fine now. 

I just can't wait until you relase your WIP.
Please excuse me in advance for deleting the spritesheet from the previous posts.  
It does not belong to me, so I think this is the right thing to do.

Congrats for the amazing work.

PS: sorry 'bout the delayed reply. Yesterday was a crazy day.

Stephane-D, 2015-06-16:
No worries ! I now these crazy days as well ;)
Also because of that i think the next SGDK version won't come that soon, i hope to make a good set of changes so that will definitely take sometime !

gilvanmaia, 2016-03-15:
Hi again! Sorry for being late on this topic.

1) Here is my e-mail address: gilvanmaia@virtual.ufc.br

2) If you would like, here is the GIMP plugin I told you about before. Each animation corresponds to a layer group, so each successive frame is stored by individual layers.

Feel free to redistribute it along SGDK:
http://tejo.virtual.ufc.br/md/tejo-spritesheet-0.3.py

Here is a sample animation image:
tejo.virtual.ufc.br/md/tejo-star.xcf

=D

Stephane-D, 2016-05-30:
Fixed with the new sprite engine.

closed 2016-05-30 by Stephane-D

gilvanmaia, 2016-05-30:
Congrats! \o/