Stephane-D/SGDK #33: tools/xgmtool/inc/gd3.h missing

aderosier, 2015-09-18:
Hello,

Was building with gendev and ran into this.

    In file included from src/xgmtool.c:8:0:
    src/../inc/vgm.h:6:17: fatal error: gd3.h: No such file or directory
     #include "gd3.h"
                     ^

The code blocks project file has a reference to this header, so just forgot to commit it I think.

Stephane-D, 2015-09-18:
Oh yeah stupid mistake, i just fixed it sorry ! Thanks for reporting !

closed 2015-09-18 by Stephane-D

aderosier, 2015-09-19:
Thanks Steph,

After getting through a couple more problems (they were gendev's fault), there is just one last thing: dma.c appears to be missing as well. This makes it so SGDK builds successfully, but games are unable to be linked to libmd.

  /opt/toolchains/gen//m68k-elf/lib/libmd.a(sys.o): In function `_vint_callback':
  sys.c:(.text+0x59a): undefined reference to `currentZ80Driver'
  sys.c:(.text+0x5c4): undefined reference to `DMA_flushQueue'
  sys.c:(.text+0x5d6): undefined reference to `DMA_flushQueue'
  /opt/toolchains/gen//m68k-elf/lib/libmd.a(tile_cache.o): In function `addToUploadQueue':
  tile_cache.c:(.text+0x92): undefined reference to `DMA_queueDma'
  /opt/toolchains/gen//m68k-elf/lib/libmd.a(sprite_eng.o): In function `SPR_clear':
  sprite_eng.c:(.text+0x424): undefined reference to `DMA_queueDma'
  /opt/toolchains/gen//m68k-elf/lib/libmd.a(sprite_eng.o): In function `SPR_update':
  sprite_eng.c:(.text+0x80a): undefined reference to `DMA_queueDma'
  collect2: error: ld returned 1 exit status

Stephane-D, 2015-09-21:
Err indeed ! It should be fixed now ! Sorry again for the troubles !