Stephane-D/SGDK #35: Error in rescomp a collision type is specified

moon-watcher, 2015-10-25:
[image]
[image]

In Rescomp, a duplicate label is generated using the SPRITE resource when the collision type is set to BOX or CIRCLE. Works if NONE or empty.

Tamk1s, 2015-10-25:
I mentioned this same problem earlier in a Spritesmind post (http://gendev.spritesmind.net/forum/viewtopic.php?f=19&t=2034#top), but good that you actually logged this as an issue

Stephane-D, 2015-10-27:
Thanks for the report, will be fixed in next version !

closed 2015-10-27 by Stephane-D