Stephane-D/SGDK #36: Split 	SPR_setAttribut() / 	SPR_setPosition() into more specific functions

nolddor, 2015-11-01:
Because you sometimes only want to change HFLIP, VFLIP,.... should be interesting create an especific funcion for each task, in addition to SPR_setAttribut()

An example could be: 

void SPR_setHFlip( Sprite *sprite, u8 HFlip )
{
	if( HFlip )
	{
		sprite->attribut |= ( 1<<11 );
	}
	else
	{
		sprite->attribut &= ~( 1<<11 );
	}
}

Others interesting functions:
SPR_setVFlip
SPR_setPalette
SPR_setPriority

The same with SPR_setPosition... 
SPR_setPositionX
SPR_setPositionY
Should be created in order to update only one axis.

Stephane-D, 2015-11-06:
I think the attributes methods are interesting but decoupling position doesn't have much interest (you rarely update only one of them).

nolddor, 2015-11-06:
It's depend on the game, in my two last games I've never update both axis at the same time... but yes I more important attributes methods than position

Stephane-D, 2015-11-06:
Hehe, you are probably developing specifics games then :p The thing is that i don't like having too much methods... I did at least the attributes methods, i will think about position... anyway i have to rework a bit this Sprite Engine stuff ;)

closed 2015-11-06 by Stephane-D