Stephane-D/SGDK #38: wavtoraw crashes

clbr, 2015-11-18:
The attached file crashes wavtoraw. It's straight exported from Audacity at 16Khz and 8-bit unsigned.

Valgrind log:
```
==4496== Command: /tmp/wavtoraw /tmp/bug.wav bug.tmp 16000
==4496==
==4496== Conditional jump or move depends on uninitialised value(s)
==4496==    at 0x400B1B: main (wavtoraw.c:68)
==4496==
==4496== Conditional jump or move depends on uninitialised value(s)
==4496==    at 0x400C36: main (wavtoraw.c:93)
==4496==
==4496== Conditional jump or move depends on uninitialised value(s)
==4496==    at 0x400C82: main (wavtoraw.c:99)
==4496==
==4496== Conditional jump or move depends on uninitialised value(s)
==4496==    at 0x400CB0: main (wavtoraw.c:105)
==4496==
==4496== Conditional jump or move depends on uninitialised value(s)
==4496==    at 0x400D70: main (wavtoraw.c:99)
==4496==
==4496== Use of uninitialised value of size 8
==4496==    at 0x400DAD: main (wavtoraw.c:129)
==4496==
```

The line numbers are offset by 2, because I had to add a #define _GNU_SOURCE on top, to fix a warning about round().

The file is named PDF because github didn't allow uploading wavs, just rename to wav.

[bug.pdf](https://github.com/Stephane-D/SGDK/files/37938/bug.pdf)

clbr, 2015-11-18:
The program is not 64-bit clean, nor endian-safe. I may be able to fix some parts.

Stephane-D, 2015-11-18:
I can't see the link with the given line (even with the offset of 2). I though char, short and long types are always fixed to 8/16/32 bits whatever is the CPU behind.

clbr, 2015-11-18:
On Wed, 18 Nov 2015 06:52:02 -0800
Stephane Dallongeville <notifications@github.com> wrote:

> I can't see the link with the given line (even with the offset of 2). I though char, short and long types are always fixed to 8/16/32 bits whatever is the CPU behind.

No, the C standard guarantees nothing of the sort. There are platforms
where char is 16 bits for example, on 64-bit Linux long is 64 bits.

Stephane-D, 2015-11-18:
oh ok. I will have to check that in the others tools as well then :-/
I don't really care about endianess currently, i guess we won't develop Genesis games on smartphone tomorrow :p

clbr, 2015-11-18:
PowerPC is big-endian, which means old Macs and some current servers. The Genesis itself is big-endian ;)

Stephane-D, 2015-11-18:
oh yeah, i forgot that old mac systems, hopefully no ones is trying to do Megadrive development on them ;)

PR #39 (2015-11-18): Wavtoraw: Linux fixes
These commits fix #38, allowing the tool to run properly on x86_64 Linux.
closed 2015-11-18 by Stephane-D