Stephane-D/SGDK #44: 4PCM_ENV driver crackles

clbr, 2015-11-27:
The attached wav file plays badly on the 4PCM_ENV driver, it crackles. The format is fine, as it's already 8-bit mono 16kHz, and it sounds perfect when player on a computer.

It plays perfect in both mplayer and aplay, the latter confirms there is no advanced filtering in play. So either there is an issue in the 4PCM driver's mixing or interpolating algorithms, or in wavtoraw.

[bug.pdf](https://github.com/Stephane-D/SGDK/files/46007/bug.pdf)

clbr, 2015-11-27:
The attachment is a gzipped wav file, renamed so github allowed upload.

Stephane-D, 2015-11-29:
Did you tested on real hardware ? I made a test using the 4PCM driver and while it doesn't sound as good than on computer (pitch is a bit lower and sound is a bit harsh) it was not crakling.
Are you doing something else while playing the sample ? You have to know than hanging the 68K bus (by using DMA or doing VDP accesses in active period) will have an impact on PCM playback. That might be your problem.

clbr, 2015-11-29:
No, I don't have real hw, just an emulator. Will test a standalone project, without any graphics, tomorrow.

Stephane-D, 2015-11-30:
Ok, you should now that PCM sound in emulator are almost time not correct. You should really test on real hardware or an accurate emulator as Exodus, BlastEm or Genesis Plus.

clbr, 2015-11-30:
A no-graphics ROM still has bad sound. I'm attaching it, can you please confirm it plays fine for you?

The source is just
```
#include <genesis.h>
#include "files.h"

int main() {

        Z80_init();
        Z80_loadDriver(Z80_DRIVER_4PCM_ENV, 1);

        SND_startPlay_4PCM_ENV(music, sizeof(music), 0, 1);

        while (1);

        return 0;
}
```

I've been using dgen. I tried to compile BlastEm, but it is seriously buggy, several compile errors. Genesis Plus doesn't exist for Linux according to Google, will try Exodus next.

clbr, 2015-11-30:
[rom.pdf](https://github.com/Stephane-D/SGDK/files/47252/rom.pdf)

Rename to rom.md.gz and unpack.

Stephane-D, 2015-11-30:
It's the almost same code i tested on real hardware, it does not play as good than on computer as i told you, but not crackling at least. DGen is a really outdated emulator, you should try newer ones as Gens or Fusion at least ;)

clbr, 2015-11-30:
Gens also didn't compile, I tried it before I tried dgen.

I'm attaching what I hear from DGen - is what you hear this bad?

(rename to .wav.gz)
[output.pdf](https://github.com/Stephane-D/SGDK/files/47277/output.pdf)

clbr, 2015-11-30:
Fusion is closed source, and Exodus was Windows only. It seems there are no other Genesis emulators I can try...

Stephane-D, 2015-11-30:
You should try this one : http://segaretro.org/Gens/GS
About the sound, it does sound better on real hardware but not too much...
I've to see why the sound is fuzzy like that, even in emulator.

clbr, 2015-11-30:
I did. All of gens, gens/gs, and gens/gs II only support 32-bit.

Stephane-D, 2015-11-30:
Did you tried that method ?
http://danielj.se/2011/05/01/install-gensgs-megadrive-emulator-on-64-bit-ubuntu/

clbr, 2015-11-30:
I run a pure 64-bit distro on purpose. It is not possible to install 32-bit libs, and even if it were, I'm not willing to do so.

clbr, 2015-11-30:
Would it be possible for you to record how it sounds on real hw? I'd like to hear if it's acceptable enough.

Stephane-D, 2015-11-30:
Ok, too bad ... i though there were more possibilities to run megadrive emulator on 64 bit Linux system :-/  You can eventually use Wine but i guess that is not a solution for you.

Stephane-D, 2015-11-30:
Unfortunately i can't as my laptop does not have a working microphone input :-/ I could eventually record it from my smartphone but the quality won't be really nice to check about quality i guess.

clbr, 2015-11-30:
Can we ask on some forum? I don't know where Genesis homebrew folks hang out. Real hw recording would confirm if there's an issue with the sound driver code.

Stephane-D, 2015-11-30:
Of course you can ask on the forum in the Megadrive or SGDK section, if you provide the rom i am sure someone could make a small record about how it sounds on the real hardware.

clbr, 2015-11-30:
http://gendev.spritesmind.net/forum/viewtopic.php?f=19&t=2256#p27812

clbr, 2015-12-02:
A kind forum member posted a recording (attached here). It still has the same kind of buzz as the emulator, just a bit quieter - this confirms there is an issue in the 4PCM driver.

[hwrec.pdf](https://github.com/Stephane-D/SGDK/files/49518/hwrec.pdf)

Stephane-D, 2015-12-02:
After having inspecting the waveform produced, the driver does not seem wrong here, hardware related issue :-/ So i close.

closed 2015-12-02 by Stephane-D