Stephane-D/SGDK #52: SGDK causes conflict with gcc stdint.h

doragasu, 2015-12-30:
I usually use data types provided by stdint.h, such as uint16_t. But including this file on projects using also SGDK, causes type redefinition problems:

```
In file included from /opt/toolchains/gen//m68k-elf/include/genesis.h:7:0,
                 from main.c:8:
/opt/toolchains/gen//m68k-elf/include/types.h:142:21: error: conflicting types for 's8'
 #define int8_t      s8
                     ^
/opt/toolchains/gen//m68k-elf/include/types.h:79:14: note: previous declaration of 's8' was here
 typedef char s8;
              ^
Makefile:148: recipe for target 'main.o' failed
make: *** [main.o] Error 1
```
I think conflicting with standard header files should be avoided. Could this be fixed?

Stephane-D, 2015-12-31:
That change was basically made for better stdint compatibility O_o :
https://github.com/Stephane-D/SGDK/pull/45

So it looks that it can bring some problems as well ?

doragasu, 2015-12-31:
Hum... I'm using SGDK on Linux built by the script by Kubilis. I built the toolchain 2 days ago, maybe the script didn't build the latest git version. I'll check if that code is present inside types.h

doragasu, 2015-12-31:
I have checked and the code is there. Once I saw the code, I immediately understood that this workaround only works if you make sure you include stdint.h before sgdk related stuff.

I reversed the include order and can confirm that now it builds without problems.

I'm closing the issue, but I suppose it would be better including stdint.h to avoid these problems.

Thanks for the support!

closed 2015-12-31 by doragasu

Stephane-D, 2016-01-02:
You're welcome :)