Stephane-D/SGDK #57: Tutorial 3 main.c uses VDP_setTileMap when I think it should be VDP_setTileMapXY

s7jones, 2016-01-30:
There are 4 arguments and I think that it needs to be the latter function?

Also will the documentation be included on Github?

s7jones, 2016-01-30:
Infact I'm having a lot of problems following Tutorial 3.

At the moment I'm stuck on trying to add the blue_moon.jpg/blue_moon.bmp.

I've ran rescomp.exe on blue_moon.bmp to make a blue_moon.res and blue_moon.h.

Then blue_moon.h is:

> extern const Bitmap moon;

But the compilation fails because

> u16 w = moon[0];

> u16 h = moon[1];

> VDP_setPalette(PAL1, (u16 *)&moon[2]);

> VDP_loadBMPTileData((u32*)&moon[18], TILE1, w / 8, h / 8, w / 8);

and Bitmap moon is a struct not an array.

So instead I do

> u16 w = moon.w;

But I start to get lost after then?

Am I on the right track?

Can I take it that the tutorials are out of date / incomplete at this point?

Stephane-D, 2016-01-30:
Ok, i fixed all wiki / tutorial stuff.
You should find updated sources files now to reflect what we have into wiki pages :)

closed 2016-01-30 by Stephane-D