Stephane-D/SGDK #69: License clarification

clbr, 2016-06-20:
http://gendev.spritesmind.net/forum/viewtopic.php?f=19&t=2464

```
Stef, I believe you intended to use LGPL to receive modifications for SGDK only. However, as is, all programs using SGDK without being LGPL-compatible are infringing.

This is because Gen programs are statically linked, and static linking is a derivative work unless otherwise noted. It would be good to clarify this, for example by adding an exception to the readme similar to FLTK's exception:
http://www.fltk.org/COPYING.php
```

jgyllinsky, 2016-08-26:
This. SDL went through similar issues (https://wiki.libsdl.org/FAQLicensing) to allow it on iOS. Any chance SGDK will ever be re-released under zlib/mit/bsd etc?

Stephane-D, 2016-08-28:
SGDK will move to the more permissive MIT license on next version :)

closed 2016-08-28 by Stephane-D