Stephane-D/SGDK #72: Java requirement in installing SGDK for lz4w.jar and WINE+Linux

jgyllinsky, 2016-09-23:
Question
---
Is there any way to make java an optional requirement?

Issue
---
Commit 6bce5c9dd14eec10b87153fd5d89875b3c15e950 introduced lz4w.jar which requires java to run. I run SGDK via WINE under linux and have had difficulty getting java to run through wine. 

Previously this was not an issue with a version of SGDK from googlecode hosting. Unfortunately since the https://sgdk.googlecode.com/ page is now down, I am unable to say the exact version that used to work under wine (~0.9x?) and Ubuntu (~10.04 LTS).

I noticed this issue while trying to install the first release of 1.22 and reproduced it with 1.22b-dev (state of SGDK at commit fafd6dde286c18a327b100e53a270d698ce03c6e). A section of the output is here:
```
Resource: BITMAP logo_lib "image/logo_lib.png" -1
--> executing plugin BITMAP...
Executing \home\user\wineopt\sgdk\SGDK-master\bin/appack c "pack.in" "pack.out" -s
Executing java -jar \home\user\wineopt\sgdk\SGDK-master\bin/lz4w.jar p "pack.in" "pack.out" -s
Can't recognize 'java -jar \home\user\wineopt\sgdk\SGDK-master\bin/lz4w.jar p "pack.in" "pack.out" -s' as an internal or external command, or batch script.
```
If this is already optional and I am not getting the option because of either my configuration or wine, how I setup my environment variables is in the "Extra info" section. 

Extra info
---
While I no longer have access to the old version of SGDK or the scripts to get it to work cleanly under WINE (batch and bash scripts), I have started new ones. I may or may not have previously tried to add to the wiki on the old google-code page, I can't recall. This is the work in progress for my NEW batch script to install as of v1.22:

```
@echo OFF

:START

	echo.
	echo.
	echo.
	echo 1) Starting:
	echo ------------
	set "str_ScriptName=%~f0"
	echo Now running %str_ScriptName%
	echo.

	setlocal

:CREATEENVVARS
	echo.
	echo 2) Creating EnvVars:
	echo --------------------
	set "str_MyCurrentDir=%~dp0"

	set "GDK=%str_MyCurrentDir%"
	set "GDK_WIN=%str_MyCurrentDir%"
	set "GDK=%GDK:\=/%"

	set "GDK=%GDK:~3%"
	set "GDK=/%GDK%"

	if %GDK:~-1%==/ set GDK=%GDK:~0,-1%
	if %GDK_WIN:~-1%==\ set GDK_WIN=%GDK_WIN:~0,-1%

	echo GDK is %GDK%
	echo GDK_WIN is %GDK_WIN%

	echo.
	mkdir "%str_MyCurrentDir%\build"

:CHECKINGPATH
	echo.
	echo 3) Checking PATH:
	echo --------------------

	set "PATHREAL=%PATH%"
	echo PATH = %PATH%
	echo.

:PATHCLEARQUESTION 
	echo Conflicting GCC setups in PATH can cause issues. 
	set /p "EMPTYPATH=Should PATH be temporarily cleared? (Y/[N])?"
	if /i "%EMPTYPATH:~,1%" EQU "Y" goto YCLEARPATH
	if /i "%EMPTYPATH:~,1%" EQU "N" goto NCLEARPATH
	echo Please type Y for Yes or N for No
	goto PATHCLEARQUESTION

:YCLEARPATH
	echo Setting PATH to just GDK stuff so that other CC1's do not conflict.

	REM Assuming that the current directory exists since that is where this file is.
	set "%PATH%=%GDK%;%GDK_WIN%"

	REM if EXIST %GDK_WIN%\bin set PATH=%PATH%;%GDK_WIN%\bin
	if EXIST %GDK_WIN%\bin set PATH=%GDK_WIN%\bin
	echo PATH = %PATH%
	goto CONTINUEAFTERCLEARQUESTION

:NCLEARPATH
	set "TPATH=%GDK%;%GDK_WIN%;%PATH%"
	set "%PATH%=%TPATH%"
	goto CONTINUEAFTERCLEARQUESTION

:CONTINUEAFTERCLEARQUESTION

	echo %GDK_WIN%\bin added to PATH

	echo.
	echo PATH = %PATH%

:LAUNCHING
	echo.
	echo 3) Launching:
	echo ------------------

	%GDK_WIN%\bin\make -f %GDK_WIN%\makelib.gen clean
	%GDK_WIN%\bin\make -f %GDK_WIN%\makelib.gen

	set "PATH=%PATHREAL%"
	echo PATH = %PATH%

:END
	endlocal
	@echo ON
	exit /b

```

To run it in wine, run `$ wine cmd` and then `> ./build.bat` , assuming that is the batchscript's name. I am running xubuntu16.04 and wine-1.9.19.

andwn, 2016-09-23:
I don't understand the necessity for wine when there is a native linux toolchain (though a bit out of date, 1.20) but maybe this will help:

https://github.com/Stephane-D/SGDK/blob/master/res/libres.res

See how the font and logo on line 1 and 2 end with "-1"? If you change them to "0" (no compression) or "1" (aplib) instead of "-1" (best/any) it shouldn't try to use java. You will have to change the same thing in your own project's res files if you had any compression enabled.

jgyllinsky, 2016-09-23:
Neat! That makes it work for me. Thanks @andwn (and of course @Stephane-D for SGDK :)).

Since I can now confirm that this batch builds SGDK, I'll submit it for pull

jgyllinsky, 2016-09-23:
Should libres.res be set to 1 by default then?

Stephane-D, 2016-09-23:
There is a linux compatible SGDK here :
https://github.com/kubilus1/gendev

Also the purpose of using java compiled tools is in fact, to have SGDK working on other platform more easily as java is available for each platform, i don't even need to provide source, just the jar file :)
The only required native part is still the GCC basic toolchain for m68k-elf target (which you can compile on a *nix type like OS).
If you want to avoid the lz4w compression, just these value in resource compression level:
- NONE (or 0) for no compression
- APLIB (or 1) for aplib compression
Avoid BEST (-1) or LZ4W (2) so it won't use lz4w.jar.

Still by default it makes sense to use -1 for library resource, normally you should have all tools available  :)

jgyllinsky, 2016-09-23:
@Stephane-D , Ok, I totally get the java reasoning - I write a lot of java myself. Actually, I was rather surprised java wasn't installing easily under wine anymore.

Is is aplib packaged as part of SGDK? Hmmm, should -1 trigger a check for java then drop to 1 etc when reading libres.res, or should I put that in the batch file? I can code either, which check location makes more sense to you?

I think my use of wine for SGDK predates gendev, and before 1.22 I wasn't sure how to handle license stuff. I guess I probably should use the native system now though...

Oh, and note that the batch script is technically for windows and not linux.

Stephane-D, 2016-09-23:
I was going to ask about the script but the .bat extension was a tip about it ;)
Well indeed i think you should try to use the linux gendev stuff:
https://github.com/kubilus1/gendev
At worst you can try to patch outdated parts from official SGDK but i think it can work almost without any changes. You can also ask Kubilus about it, i think he's ready to help about get it working. It's really a shame that you have to use Wine for SGDK.

kubilus1, 2016-09-23:
Hello, maintainer of the linux gendev stuff here, attempting to upgrade to the latest sgdk I am also bit by the Java required bug.  

If the purpose is to have a more platform neutral scripting language, I would strongly suggest using something like Python instead of Java.  Java is very heavy weight, and may not be available by default on most distros.  That's my 2cents on it.  

What's cool about Python from a platform independent scripting language is available for other users to expand on tools and learn.

-----

Making lz4w.jar available in the correct location and installing Java works, so that's not a big deal though, really.

----

Attempting to upgrade to the latest SGDK, I get errors attempting to compile the sample programs.  These look to me like incompatibilities between the new libs and the old examples.  It's probably on my end, but do we know for sure that all the example programs compile okay?

andwn, 2016-09-23:
How are you building the samples and what error? I can try with my fork. I'm unaware if the samples compile but at least my project does.

Using the new sega.s?

Stephane-D, 2016-09-23:
The things about python is that:
- i don't know have a single experience with this language so for me java is easier (even if python should be easy to learn) :p
- python is, imo, mainly linux friendly but not that much for Windows or OSX operating system (you have to install it separately) but i do agree that python is much lighter than java.
- java is required for many applications today so almost of time it's already installed on your system.
- java is much faster and this can be definitely interesting for image processing / compression tools (as SGDK can embed).
- java can do native GUI application without installing extra GTK lib stuff or so...

To be honest java or python, for me both are not ideal... I would have preferred to stay with pure C but outch ! It really hurts to use pure C for complex tools (rescomp and xgmtools become more and more complex and i think i would have been *much faster* using java from beginning).

If you experience about demo/sample compilation, i can try to help :)

jgyllinsky, 2016-09-23:
@kubilus1 under the setup described above, I build  6 examples successfully. They are all named out.bin in their respective directories, but renamed they are :
   1. cube_flat.bin
   2. hs_effect.bin
   3. joytest.bin
   4. partic.bin
   5. sound.bin
   6. sprite.bin

I don't think "bench" builds, or at least there is no` /out` folder .

As for the requirement for java, I build fine with setting that value to 1, so everything is OK on my end :+1: Python vs java vs C ... ah languages. :p

@Stephane-D native GUI application.. generically or future SGDK plans?

Stephane-D, 2016-09-23:
@jgyllinsky Well, who know :) At least the possibility remains open (it may be useful for resource compilation) and i am already using it to debug some of my tools.

jgyllinsky, 2016-09-24:
Oh, and just in case someone reads this later, I changed "`SPRITE enemies_sprite "sprite/enemies.png" 6 4 FAST 5`" to "`SPRITE enemies_sprite "sprite/enemies.png" 6 4 0 5`" in `sprite.res` files. I did this since the wiki page says that the format for sprites.res is "`SPRITE output_var_name "gfx/file.bmp" <sprite_width> <sprite_height> <compression> <time>`".

When making the out.bin file, this batch script is run in the sample's root directory and the script itself is generated by the updated build.bat script:

```
set GDK=/home/user/wineopt/sgdk/SGDK-master
GDK_WIN=Z:\home\user\wineopt\sgdk\SGDK-master
PATH=/home/user/wineopt/sgdk/SGDK-master;Z:\home\user\wineopt\sgdk\SGDK-master;Z:\home\user\wineopt\sgdk\SGDK-master\bin
Z:\home\user\wineopt\sgdk\SGDK-master\bin\make -f Z:\home\user\wineopt\sgdk\SGDK-master\makefile.gen
```

Stephane-D, 2016-09-26:
In fact for compression field now you can set it to -1/0/1/2 or BEST/NONE/APLIB/LZ4W or FAST

Stephane-D, 2018-04-04:
Closing this one as future SGDK may rely even more on java being installed (will become mandatory at some point) so the linux gendev script should add a dependency to java if not already installed.

Stephane-D, 2018-04-04:
Maybe better to wait for @kubilus1 to close this one ;)

Stephane-D, 2018-04-04:
Closing this one as future SGDK may rely even more on java being installed (will become mandatory at some point) so the linux gendev script should add a dependency to java if not already installed (issue should be on gendev side).

closed 2018-04-04 by Stephane-D