Stephane-D/SGDK #75: Illegal instruction! when testing rom

salvadorc17, 2016-12-12:
Hi, i have managed to compile a rom using CodeBlocks, and following tutorial, but when want to test rom in any emulator got "Illegal instruction" message

Stephane-D, 2016-12-13:
Hi Salvador,
I guess there is something really wrong with your code then :-/ What is your code ?

salvadorc17, 2016-12-13:
Not really, is the hello world tutorial code, how can be wrong

Stephane-D, 2016-12-14:
Oh indeed there is something really wrong with the development environment then...
Is it possible that several GCC compilers are conflicting together so you end up with the wrong GCC being used to compile the rom ? The test rom you are using is rom.bin file ?

salvadorc17, 2016-12-14:
Yes, it is, also i dont have multiple GCC compiler only this one used for SGDK, unless is problem with Code Blocks?

Stephane-D, 2016-12-15:
Can you try a build rebuild (using rebuild or clean/build) and send the compilation log ?

salvadorc17, 2016-12-15:
Ok, thanks i found the problem was missing the while statement.

Stephane-D, 2016-12-16:
Oh ok, indeed you need it otherwise program end by executing outside of range code...
I can close the issue now :)

closed 2016-12-16 by Stephane-D