Stephane-D/SGDK #76: Better tutorials.. how to draw tiles and sprites?

salvadorc17, 2016-12-19:
Im trying to understand how to draw tiles and sprites properly but the tutorials are outdated, and some methods used there are deprecated, can you help me telling how to achieve this..

Stephane-D, 2016-12-20:
The best tutorial are the included samples in SGDK, the 'sprite' sample is a good example about how doing basic operations :)

Stephane-D, 2016-12-20:
Duplicate of #5

closed 2016-12-20 by Stephane-D

salvadorc17, 2016-12-20:
Still outdated, some methods there have wrong arguments and other does not work, you need to update those also..

salvadorc17, 2016-12-20:
Can you tell me why using spr engine does not work properly, when using SPR_update() this give weird colors to image and do not draw sprite.

Stephane-D, 2016-12-21:
There is something probably wrong somewhere in the code.
Did you have uploaded the palette manually before ? Did you correctly initialized the Sprite engine ?
Use http://gendev.spritesmind.net/forum/ forum to have support, you will find tons of help / answer about SGDK use :)

salvadorc17, 2016-12-21:
You mean vdp_setpallete and spr_init(), yes i have used those, the bitmaps draw properly, but when using spr_update the thing got broken.

s7jones, 2017-02-06:
I've not been able to adapt my old files based on the outdated tutorial to the new changes to SGDK. I'd be interested in a redone tutorial too. Thanks.

Stephane-D, 2017-02-07:
I will try to definitely update them on next big SGDK update !
Unfortunately i'm really busy both with my professional and private life now, as soon i've a bit of spare time i will get back working on SGDK :)