Stephane-D/SGDK #78: Auto-build script

chrilith, 2017-01-09:
Hello,

Here is a small batch to automatically build the library in both release and debug mode whitout defining the environment variables.

```
SET "GDK=%cd:\=/%"
SET "GDK_WIN=%cd%"
SET PATH=%GDK_WIN%\bin;%PATH%

make -f %GDK_WIN%\makelib.gen clean
make -f %GDK_WIN%\makelib.gen release
make -f %GDK_WIN%\makelib.gen debug

@ECHO.
@ECHO.
@ECHO -------------------------------------------
@ECHO SGDK is now ready! Press any key to exit...
@PAUSE >nul
```

The `build.bat` file must be in the same directory as `makelib.gen`.

Chris

bu95bunny, 2017-01-17:
Nice 1, thank you!
> The build.bat file must be in the same directory as makelib.gen.
Or just change the path to (wherever it is)\makelib.gen

Stephane-D, 2017-01-18:
I added the file as build_lib.bat in root folder of SGDK, thanks :)

closed 2017-01-18 by Stephane-D

jgyllinsky, 2017-02-28:
Issue linked with pull request #73 , Create build.bat (local build script), from Sep 23, 2016.