Stephane-D/SGDK #79: 3D features

lab313ru, 2017-02-14:
It would be great to add some basic 3D functionality. Polygons, textures, vectors, etc.

For example, Wolfenstein 3D, or Zero Tolerance. Both of them are using some sort of 3d-computing functionality, so, adding this to SGDK will help to build some great projects!

Stephane-D, 2017-02-14:
There is already some basics 3D functionality in SGDK : 
https://github.com/Stephane-D/SGDK/blob/master/sample/cube_flat/src/main.c

You have complete 3D transformation support with this unit ;
https://github.com/Stephane-D/SGDK/blob/master/inc/maths3D.h
And polygon drawing using the Bitmap engine with this method (unit bmp) : 
u16 BMP_drawPolygon(const Vect2D_s16 *pts, u16 num, u8 col)

Still there is no raytracing engine as this is quite specific and there is many ways of doing it...
But if someone make a good engine i will be happy to integrate it ;)

lab313ru, 2017-02-15:
What about a drawing speed/how much interlacing applied?

abritinthebay, 2017-02-15:
That would depend on whatever engine _you_ write with the code...

Stephane-D, 2017-02-16:
Not sure about what you mean by interlacing. Again you speak about the very specific case of raycasting used in wolf 3D but there is many ways of doing 3D on Megadrive ;)
The 'bench' sample contains a polygon drawing performance test if you are curious ;)

Stephane-D, 2017-04-27:
I'm closing this issue because definitely too specific for a general purpose SDK.
There is already basic 3D methods for flat rendering, having a raycaster could be a nice addition but more as an optional extension imo :)

closed 2017-04-27 by Stephane-D