Stephane-D/SGDK #81: There is no differences between libmd release and debug one

lab313ru, 2017-06-20:
Subj.

Stephane-D, 2017-06-21:
What do you mean exactly by "no difference" ? At least the library size is not the same. Also now debug target use optimization level 1 (while release use O3) so performance even in debug are already good.

lab313ru, 2017-06-21:
I mean, that after compilation on Windows, libmd.a, and libmd_debug.a have the same sizes.
And my tries to change compilation flags don't change that difference.

lab313ru, 2017-06-21:
This is a compilation log.
[build_lib.zip](https://github.com/Stephane-D/SGDK/files/1090894/build_lib.zip)

Stephane-D, 2017-06-21:
Oh yeah you need to do a clean between release / debug. Maybe something to change in the makefile to avoid that...

Stephane-D, 2017-06-21:
I fixed the build script ;)

closed 2017-06-21 by Stephane-D