Stephane-D/SGDK #83: VBCC usage?

lab313ru, 2017-06-27:
If you know, it is a very optimizing compiler/assembler. It knows about M68K, and even Z80.
It will be great to move SGDK to it.

Stephane-D, 2017-06-28:
If you can convince me that VBCC is better then i can consider the proposition but I know this compiler. I was interested with VBCC for sometime then i made some benchmarks with it and i was not really satisfied with the generated code (worse than the old GCC 3.4.6 in almost case). Today SGDK is using GCC 6.3 and code generation is much better than GCC 3.4.6 so i think there is even less reasons now to move on VBCC :-/  Why do you prefer VBCC ?

lab313ru, 2017-06-28:
It has many optimizations methods. And code looks really optimized) Maybe since that time when you've tested it, it was much more optimized.

Stephane-D, 2017-07-19:
We should compare them then :) Maybe it's more optimized now but GCC 6.3 bring an important speed boost as well. Also GCC is pretty "standard" and common...

Stephane-D, 2017-09-24:
Thinking about it, almost certain VBCC won't be a good change for SGDK. Of course anyone can choose to use VBCC (using fork of SGDK eventually) but by default i prefer to stay with GCC, even more now we're back using recent version of it (maybe we can switch to GCC 7 in a close future).

closed 2017-09-24 by Stephane-D