Stephane-D/SGDK #85: Wrong alignment for data?

lab313ru, 2017-06-28:
- smp_null.s - align 256
- smp_null_pcm.s - align 256

```
    addr = (u32) smp_null;
    // sample address (256 bytes aligned)
    pb[0] = addr >> 8;
    pb[1] = addr >> 16;
    // sample length (256 bytes aligned)
    pb[2] = sizeof(smp_null) >> 8;
    pb[3] = sizeof(smp_null) >> 16;
```

but:
```
    addr = (u32) smp_null_pcm;
    // sample address (128 bytes aligned)
    pb[0] = addr >> 7;
    pb[1] = addr >> 15;
    // sample length (128 bytes aligned)
    pb[2] = sizeof(smp_null_pcm) >> 7;
    pb[3] = sizeof(smp_null_pcm) >> 15;
```

Stephane-D, 2017-06-28:
Not really important is it :p ? It is better to have higher alignment here as i am not sure the Z80 driver will actually support it...

lab313ru, 2017-06-28:
I mean, that in code you are using alignment by 128 (and doing shift by it). But data is actually aligned by 256. Is it mistake?

And, my compiler tells me about alignment is too big. That's how I found this place.

Stephane-D, 2017-06-29:
No it is not a mistake, it's a big alignment indeed but it's required by the Z80 driver :) Even if the 2ADPCM could eventually have 128 bytes aligned sample in theory i believe it can lead to some issues sometime, so having 256 bytes alignment is safe here.
VBCC tells you about the alignment problem i guess ? GCC doesn't complain about it, i hope VBCC respect it though otherwise you will have troubles with Z80 sound drivers.

closed 2017-06-29 by Stephane-D