Stephane-D/SGDK #90: No support for compressed VGMs?

minerscale, 2017-07-29:
I can't help but notice that there is no setting for compression for VGM files. My VGM file is over 40KB uncompressed and as a .vgz it is at a more manageable 4 KB. Is there any reason this hasn't been done yet? If not? This would be a pretty nice feature. Cheers!

LukeUsher, 2017-07-29:
I think the megadrive doesn't have enough processing power to decode compressed VGM, additionally there is only 64kb of ram, so space is an issue too.

If you want smaller music data, use a real sound engine such as ECHO. If you just want to make a VGM player, uncompressed is the only way.

minerscale, 2017-07-29:
Oh, ok. So for a soundtrack for my game, I'll have to go with ECHO then.

Thanks a lot for tipping me off to that. But it's a damn right shame composing music for the genesis just got much harder.

closed 2017-07-29 by minerscale

andwn, 2017-07-30:
Could always give dmf2esf a spin http://gendev.spritesmind.net/forum/viewtopic.php?t=2245

I've not used it myself but it sounds more like something you are looking for than xm.

Stephane-D, 2017-07-31:
Note that your VGM is optimized and converted to XGM format which is lighter than VGM file, still it won't be as small than a compressed VGM file but you may test the final size just by converting your VGM to XGM file using the xgmtool :)
If you want to have very small footprint for your music Echo may be the best option then :)

minerscale, 2017-08-01:
Alright cool :) I'm already using echo now. But the lack of tools is a bit of a bummer : | dmf2esf isn't perfect. But I can fix it up.