Stephane-D/SGDK #91: Make debug working extremely strange...

minerscale, 2017-08-06:
I don't know what the problem is but by game builds and runs as expected when I make normally but if I make debug parts of my tiles are not there... Not sure what the problem is, I'm using gendev, if it makes a difference.
[rom.zip](https://github.com/Stephane-D/SGDK/files/1202811/rom.zip)

[image] - made with debug
[image] - made with release

Stephane-D, 2017-08-10:
Is it fixed or not ?

minerscale, 2017-08-10:
No, it's not fixed... I just labelled  the images wrong... :p

Stephane-D, 2017-08-10:
The thing is that without your project source code i'm not able to reproduce this problem... i never observed it and i'm using *a lot* the debug target so i really need your project folder to understand what is going wrong there :)

minerscale, 2017-08-11:
Oh... ok then. I was keeping it closed source for the moment. But, whatever. It's so early in development that it doesn't matter. I was in the middle of testing the level system so when you press start it changes level \o/

Stephane-D, 2017-08-11:
If you feel more comfortable with you can just distribute a minimal example which can reproduce the issue but i think that is the whole which make the problem to appears ;)

minerscale, 2017-08-11:
No, it's fine. I'll just send it over :)

Stephane-D, 2017-08-11:
Ok thanks i will try to reproduce the issue here then :)

minerscale, 2017-08-11:
cool :)

Do you have any insight of what's going on?

Stephane-D, 2017-08-11:
Absolutely no idea for now to be honest... debug and release should produce the same result, the only difference is about speed but i don't see how it can generate different "visible" results..

minerscale, 2017-08-11:
huh, alright then. Hope you find something.

Stephane-D, 2017-08-11:
Do you observe the bug only on real hardware ? Testing quickly with Fusion i can't see any differences with debug and release builds. Also i have only 2 levels, first have yellow block while second has some weird characters blocks, is it expected ?

minerscale, 2017-08-12:
Yeah, the weird character blocks are expected (Test sprites lol). Believe it or not, I don't actually have a Mega Drive to test with yet... (yet). It's working fine for me too... Must be quite selective... This is a different build of the game to what the original bug report, however there might be a build in the git repo that the bug is reproducible in. Yeah there is, just checked... Here's the reproducible build. Crazy how much it's changed in just 6 days! The already built rom is the erroneous rom built with debug.
[genesis-rpg.zip](https://github.com/Stephane-D/SGDK/files/1219719/genesis-rpg.zip)

Stephane-D, 2017-08-16:
Ok, i had a quick look on your sources ans realized you don't protect VDP access with SYS_disableInts() / SYS_enableInts() couple. It's important to protect any VDP accesses (as VDP_setTilemapXY(..)) otherwise you can screw up the VDP state because SGDK V-Interrupt callback will access VDP too.
Adding the SYS_disableInts() / SYS_enableInts() where you need it fix the issue.

closed 2017-08-16 by Stephane-D

minerscale, 2017-08-17:
Huh, ok then. Good to know. I'll add that now. Might be a good idea to put that somewhere in the wiki :p