Stephane-D/SGDK #92: Sprite Engine complex animation

dimiro1, 2017-08-09:
Is it possible set the duration of a single sprite animation with the rescomp utility?

```
A(5) B(5) C(1)
D(8) E(5) F(5)
```

As I am understanding, now, it is possible to set only a single time value:

```
SPRITE mysprite_sprite "sprites/mysprite.png" 6 6 0 5 NONE
```

Stephane-D, 2017-08-10:
Unfortunately right now you can indeed only setup the duration for the whole animation.
I need to improve the resource form (using XML probably) to allow that.
What you can eventually try is to modify the generated .s file from rescomp (using hand compilation) so you change change duration for each frame but as soon you will modify your resource, the makefile will regenerate the .s file so that is a really convenient solution :-/
A proper Sprite editor with proper XML format would be the solution...
I keep the issue open in the meantime...

dimiro1, 2017-08-10:
Thanks for your reply.

For the "array" parameter I am thinking in something like this:

```
SPRITE mysprite_sprite "sprites/mysprite.png" 6 6 0 "5,5,1|8,5,5" NONE
```

What do you think?

Stephane-D, 2017-08-10:
I would probably do something as usual array format in programming language :
[[5,5,1][8,5,5]]
without any space obviously as space are used to separate parameters ;)

dimiro1, 2017-08-10:
Do you want to keep retro compatibility? I can see two approaches:

1. Change the command for SPRITE_ANIMATION or SPRITE_ARRAY
2. Add an extra parameter at the end of the command.

What do you think?

Stephane-D, 2017-08-10:
Of course i cant to keep retro compatibility :)
If i detect '[' first character then i assume an array otherwise an assume a single value

dimiro1, 2017-08-10:
@Stephane-D Thanks, I am not familiar with the codebase and I am working on OSX, but, I will try to implement it.

Stephane-D, 2017-08-10:
Oh you plan to do the code change by yourself ? that would be really handy ! thanks !

dimiro1, 2017-08-10:
I hope soon open the PR. :)

Stephane-D, 2017-08-10:
Cool :) One suggestion, if the array does not match the number of expected row/column, you may just display a warning about it and use the last found value for remaining frames.

Stephane-D, 2023-12-02:
> I look at the issue from 2017 year )))
> We solved the problem in other way. In the game code.
> 
> I just want to remind that there is also unused
> **anim->loop** in SGDK sprite struct
> 
> I mean it is not possible to define it via RESCOMP params, yet.

Yeah I realized I had this information not properly used (It could even be used to avoid loop afaik). Need to change that too but the SPRITE resource declaration is already overwhelmed by parameters, I definitely need to update the resource declaration format (for something as JSON for instance)

Stephane-D, 2025-06-19:
Added support to define animation duration in an array format (see rescomp.txt for more information about the format)