Stephane-D/SGDK #94: Does the XGM driver support a single sample table for all songs?

minerscale, 2017-08-19:
I need to keep the space taken up by songs in my game to a minimum. A single table of samples used by all songs would do this. Does the XGM driver support it? Or could it be modified to do that?

Stephane-D, 2017-08-21:
Unfortunately XGM does not support single sample table right now, that was one of my first concern about the format, it could be worked up but still a lot of work is needed (need to compile all XGM resources at same time and separate sample data from music data). As my concern was to maintain the XGM conversion straightforward i couldn't add that easily.
Fading (volume control to make it short) isn't supported by XGM driver, and this feature would require an important rewrite of the driver :-/
This feature is in my todo list for XGM v2 but did not had chance to work on it a lot yet...

minerscale, 2017-08-21:
Aww, ok... ;_;

Nevermind. Still a super cool driver though! How finished is XGM v2? That seems pretty cool!

Stephane-D, 2017-08-21:
XGM v2 is just on paper :p

minerscale, 2017-08-21:
aww ;-;

stephanedamo, 2021-07-25:
Instead of playing raw ym/psg register dumps (which take quite a bit of space on the cartridge), would it be possible to develop a custom tracker format + driver to allow for super compact sizes ? A bit like what SEGA dit with its SMPS driver, if I remember correctly most Sonic songs fits in 1-2Kb and all use shared FM/PSG/DAC instruments. Also they use the same note/effect commands to do sound effects using FM synthesis which is quite cool. If the only thing preventing this from happening is the lack of tracker software to support it, then I can create it : ) (but I can't write a z80 driver in assembly, at least for now...)

Stephane-D, 2021-07-26:
There are some *huge* advantages of using register dumps format :
- can import data from VGM files (which all tracker or even emulator support)
- simpler parsing and so light CPU resource decoding, leaving time to handle more software PCM channels

And imo just that 2 points make registers dump much more interesting than custom tracker format.
Still I plan to write a new sound driver roughly based on XGM driver but greatly optimized for size in mind, even if today we don't really care about music data size honestly, but just for the sake of authenticity and also because it's kind of a bit cheaty when you don't make effort about space footprint of your data.
I don't know yet if i will add support for FM sound effect as that make the sound driver much more complicated in my case (injecting SFX in music data stream along the FM channel allocation/reservation stuff) and because honestly imo except in rare situation i think FM SFX doesn't sound great and some games really suffered from that. I know it would still be nice to have the choice but i have yet to study the tradeoff in implementing that.

stephanedamo, 2021-07-27:
Some of the best FM SFX are from Sonic 3D and Sonic & Knuckles I think, but a lot of games weren't nearly as good I agree. It's still useful to save space to stay below the 4MB limit (after that you have to add external mapper chips for a physical cartridge release?). But yes the number of people planning to design FM SFX is probably very low, so I totally understand. I may try my own things and if that works well enough I'll share it with you all !

Stephane-D, 2021-07-27:
That sounds like a very fair proposition 👍

breakthetargets, 2021-08-07:
> I don't know yet if i will add support for FM sound effect as that make the sound driver much more complicated in my case (injecting SFX in music data stream along the FM channel [...]

Personally, I think FM SFX is very ideal in order to be able to keep space size down, just because memory doesn't cost much doesn't mean we should ignore it entirely especially when 4MB is the limit without a mapper. I understand that you personally don't like it, but it's something that can sound good in the right hands. The main reason why you don't hear people make them at all is because the lack of actual need for FM SFX as XGM doesn't support it at all currently. As someone who has made music and sounds for XGM before on a project, I would have easily opted for FM SFX because the PCM SFX just took up too much space which required finessing from the devs to begin with to make sure things were able to fit because most of the space was taken up by the sound, both music and SFX.

In the end, FM SFX is something that I think should be supported as it would allow more space to be free for the developers, or even more music.

SavagedRegime, 2021-08-07:
In my experience the various sources of file size bloat from XGM was pretty challenging to manage for full scale game development. Working on Xeno Crisis, only having PCM available for SFX racked up the ROM space quite quickly when you need to make a whole game's worth of sounds and I went through several passes optimizing them the only way I could by shortening the length as much as possible while still having them sound somewhat decent and satisfying. Also had to spend time figuring out tricks to further optimize the music to bring the size down (shaved off nearly 100kb in one instance).
That was a 4 megabyte ROM and in that case I was fortunate to have roughly half of that space dedicated to music/audio (which I imagine would never had flied back in the 90s), though I doubt that kind of privilege would be as common when it comes to audio for other games in development. Just using mappers for larger ROMs as a crutch for all this is also not as easy as it sounds on the surface since not only does it complicate the development process but also risks potential incompatibilities with different emulators and/or hardware clones from what I've heard

Another thing with the lack of FM SFX support is also that you can't do layered stereo sound effects with the PCM channel. Those were common enough to add extra depth back in the days like the crumbling cliffs from Sonic.

If I could have FM/PSG SFX mixing at the cost of, say, only 2 PCM channels then I would easily take that.

stephanedamo, 2021-08-08:
Agree ! In the right hands FM is awesome. Maybe that could be done by playing small VGM files restricted to some channels as SFX, interrupting the corresponding channels of the background music.
Best would be a common FM/PSG/PCM instrument bank, with song data and sfx data containing commands rather than register dumps. I already have an YM2612 tracker prototype that can work on "projects", each project contains songs, sfx, instruments and everything is shared to ensure there is no duplicated data. Also it's possible to pack note and timing data into a single byte, providing good compression ratio on most songs. Just requires a more complex parser running on z80.

breakthetargets, 2021-08-08:
> If the only thing preventing this from happening is the lack of tracker software to support it, then I can create it : ) (but I can't write a z80 driver in assembly, at least for now...)

Late, but there's currently a tracker in the works called [ZorroTracker](https://github.com/ZorroTrackerDev/ZorroTracker) which is built so it relies on drivers to handle a ton of the back end. Currently, XGM is going to be the first driver supported, but there will be a driver made called ZorroDriver that will be built around being able to do tracker effects. Eventually other drivers, such as [AMPS](https://github.com/NatsumiFox/AMPS), which is created by the main developer @NatsumiFox, will be supported.

stephanedamo, 2021-08-08:
Cool I'll check it, thanks ! :)

freq-mod, 2021-08-11:
I can agree with @breakthetargets and @SavagedRegime - FM SFX can be very impressive in right hands, they don't weight tens of kilobytes; and I wouldn't mind having less PCM chans. 2 would be plenty enough, at least for my needs.

RaijinXBlade, 2021-08-11:
In my opinion, you should have a 1 and/or 2 PCM mode for XGM as well. You could bump up the max sample rate for those modes too, but more importantly, you could add support for sample offsets, and perhaps something like Skitchin' did, and have offset loops and pitch support as well. I'd take less PCM channels with more capability and FM/PSG sound effects any day.

Stephane-D, 2025-06-19:
Closing it as this is now supported in XGM2.

closed 2025-06-19 by Stephane-D