Stephane-D/SGDK #95: SGDK with Netbeans

shingosama, 2017-08-21:
Hi Stephane, I would like to use SGDK with my favorite IDE: Netbeans. I tried to configure it by following the logic of CodeBlock, but impossible to compile anything. Could you explain how to configure Netbeans with SGDK or do a little tutorial?

Thank you

Stephane-D, 2017-09-10:
I just spent a whole day trying to configure SGDK with NetBeans without any success :-/
If someone is able to manage it (without installing externals and useless apps as msys) then i would be really really graceful :)

moon-watcher, 2021-03-30:
Hi. I was able to compile using a simple C project and this configuration

[image]
[image]

Stephane-D, 2021-03-30:
Thanks ! Good to know we can do it that way :)