Stephane-D/SGDK #96: Plan_window doesnt run properly.

turiaso, 2017-09-03:
Hi, I am building a Sega rom with this sdk. 

I am trying set a text or a image on  windows plain but this don't show nothing.  When I paint on plain B or plain A thesis run properly.  I am using the 1.31 version. 

REGARDS

andwn, 2017-09-03:
Are you setting the Window to be visible (VDP_setWindowVPos/HPos)? Before you do this, be aware the Window Plane will hide Plane A.

Stephane-D, 2017-09-04:
As andwn said, you first need to enable the window (and be sure enough vram is allocated for its tilemap) before using the window plan.

adglow-jaruiz, 2017-09-04:
Hi,

At first congratulate you by this proyect. Amazing.

I will try these this nigth. I will response you with results.

I only need use Plane Windows to paint texts as scores and some traces at the moment. Will hide this the plan A?

Thanks you very much!

Stephane-D, 2017-09-04:
Thanks for the comment :)
Indeed window and plan A are mutually exclusive, see the window plan as a "fixed part" of plan A.

turiaso, 2017-09-04:
All clear.

Thanks you very much Stephane.

By other hand, I would like collaborate with you in this proyect and to
learn more about megadrive system.

My email is juan.ruizpachon@gmail.com, if you want add to me on hanouts to
speak more about this.

I will create 3 games and I will export them to Everdrive and run on my
sega megadrive:
- pong4
- tanks
- learn to play guitar on sega. I am developing a new controller based on
6button-pad, a sega guitar.

Is very possible that I need to ask you more times about images and sound
creation.

Regards!

:

closed 2017-09-05 by Stephane-D