Stephane-D/SGDK #99: The 0th sprite in the sprites array gets initialised on the 1st sprite according to Kmod

minerscale, 2017-10-04:
I am using this code to initialise my player sprite in the 0th position:

`sprites[0] = SPR_addSprite(&player_sprite, 0, 0, TILE_ATTR(PAL2, FALSE, FALSE, FALSE));`

But, according to Kmod, it's in the first position:
[image]

I think this is a bug, and if it isn't, could someone explain what's going on here? This shouldn't be happening if I'm correct.

closed 2017-10-05 by minerscale

Stephane-D, 2017-10-08:
Glad you fixed this one :p It looked like a complicated one X'D

minerscale, 2017-10-09:
Oh my god it was the worst thing ever, now I have to optimise it. It's too laggy ;_;

Stephane-D, 2017-10-09:
Good luck ;) Note that sprite #0 is reserved by the sprite engine ans so it's wasted (because of sprite sorting feature), so you can never use more than 79 (H40) or 63 (H32) sprites...

minerscale, 2017-10-09:
;-;