Stephane-D/SGDK #100: Strings

Kintobor, 2017-11-28:
I don't understand how strings work. I can make and add a string to the screen, but I can't take a single character from that string.

`char greeting[6] = {'H', 'e', 'l', 'l', 'o', '\0'};`
`        VDP_drawTextBG(PLAN_A, greeting, CHATBOXx + typewriterx, CHATBOXy + typewritery);`

That gives me:
[image]

`VDP_drawTextBG(PLAN_A, greeting[4], CHATBOXx + typewriterx, CHATBOXy + typewritery);`

Changing greeting to greeting[4] gives me:
[image]

`VDP_drawTextBG(PLAN_A, greeting[typewritery], CHATBOXx + typewriterx, CHATBOXy + typewritery);`

Finally, giving greeting a changing index gives me:
[image]

Stephane-D, 2017-11-28:
Usually i would something as : 
```
char greeting[6] = "Hello";
VDP_drawTextBG(PLAN_A, greeting, x, y);
```

But you can also do :
```
char greeting[16];
strcpy(greeting, "Hello");
VDP_drawTextBG(PLAN_A, greeting, x, y);

```

Still your method should work, i guess you get memory corruption somewhere else leading to that problem...

Kintobor, 2017-11-28:
I put the following code into main() of the joystick example. one_c outputs as 4 instead of n.

`    char one[] = "One";
    VDP_drawTextBG(PLAN_A, one, 1, 21);
    char one_c = one[1];
    //one_c should now be "n"
    VDP_drawTextBG(PLAN_A, one_c, 10, 21);`

[image]

Kintobor, 2017-11-28:
Doesn't work:
`if (one_c == "n") {VDP_drawTextBG(PLAN_A, "It works!", 1, 22);}`

Works:
`if (one_c == 110) {VDP_drawTextBG(PLAN_A, "ASCII Index", 1, 24);}`

Looks like I need to figure out how to convert ASCII index to a character.

Stephane-D, 2017-11-28:
You cannot do that, you are mixing String (array of char) and single char...
The VDP_drawTextBG(...) method expect an array of char, you can't write a single character only _char_ argument. If you want to draw a single character then you have to copy it inside a string : 

```
char one[] = "One";
char one_c[2];

one_c[0] = one[1];
// always terminate string with null character
one_c[1] = 0;          // equivalent to '\0'
// one_c should now be "n"

VDP_drawTextBG(PLAN_A, one, 1, 21);
VDP_drawTextBG(PLAN_A, one_c, 10, 21);
```

Also you can't do that : 
`if (one_c == "n") ...`
To test single character you have to do that : 
`if (one_c[0] == 'n') ...`

Your bug are just coming from that so i guess i can close this issue ;)

closed 2017-11-28 by Stephane-D

Kintobor, 2017-11-28:
Of course! Everything makes sense now! I shall make a secret shrine to you in my game.

Stephane-D, 2017-11-28:
Hehe ^^ You don't have to don't worry ;)