Stephane-D/SGDK #101: Customizable sprite loop frames

heitorgcosta, 2017-12-11:
Hi Stephane,
I've recently discovered the SGDK and I'm really loving it.
I was doing some tests today with some shmup-like prototype when I've come across with something: The "loop" property of the SpriteDefinition is always zero, so all animations will loop back to frame 0.

I've made some "homemade" changes to the Sprite plugin in the rescomp to add customizable frames to loop for each animation, with an array-like notation:

`SPRITE styx_sprite "sprite/styx.png" 4 3 FAST 5 NONE [0,1,1]`

The first animation will loop back at 0, while the other two will loop back to 1 (which in this case, is "no-loop"). The tests was successful.

With that in mind, I wanted to know if, after I refine the solution (which is a bit messy at the moment, hehe), I can open a PR.

Also, doing that I discovered that the `SPR_setAnim()` and `SPR_setAnimAndFrame()` didn't reset the frame's timer, leading to, sometimes, the first frame being sort of skipped. I also made a fix for this if you don't mind.

Thank you for listening.
And thank you very much for the SGDK! It's awesome! :3

Stephane-D, 2017-12-11:
Hi,

Thanks for your very nice comment about SGDK =) I'm glad you appreciate it !
Your addition is very interesting and in fact there is already a open issue about doing that as well for the animation time information ;)

About SPR_setAnim(..) and SPR_setAniyou're right, they should set timer to 0, thanks for reporting the issue ! I can fix it to on my side :)

D0NM, 2022-12-21:
Hm... imo it is not flexible.
We made a tiny wrapper that does all the needed stuff. So other coders did the same.

Stephane-D, 2022-12-28:
> 

Indeed i'm planning since some time to update the resource file format using something as json so we can easily set advanced properties from some resource definition (SPRITE mainly)