Stephane-D/SGDK #103: Critical issue: sprite artifacts spilling on background planes?

timotei, 2017-12-22:
Hello,

It seems starting with version 1.32 there is a big issue. The following piece of code: 

```
#include <genesis.h>
#include "resources.h"
int main()
{
	VDP_setPlanSize(128, 32);
	VDP_drawImage(PLAN_B, &background, 0, 4);

	VDP_setHorizontalScroll(PLAN_B, 0);
	VDP_setVerticalScroll(PLAN_B, 32);
	SPR_init(0, 0, 0);

	VDP_setPalette(PAL2, sonic_sprite.palette->data);
	Sprite *s1 = SPR_addSprite(&sonic_sprite, 100, 128, TILE_ATTR(PAL2, 0, 0, 0));
	SPR_setAnim(s1, 2);
	Sprite *s2 = SPR_addSprite(&sonic_sprite, 150, 128, TILE_ATTR(PAL2, 0, 0, 0));
	SPR_setAnim(s2, 3);
	Sprite *s3 = SPR_addSprite(&sonic_sprite, 200, 128, TILE_ATTR(PAL2, 0, 0, 0));
	SPR_setAnim(s3, 7);

	while(1)
	{
		SPR_update();
		VDP_waitVSync();
	}
	return (0);
}
```

yields the following rendering:
[image]

* That code works fine on the 1.22, 1.30 or 1.31 version of SGDK
* The background and sonic sprites are from the sprite sample of SGDK
* The more sprites one adds, the more artifacts will be written there.
* Plane A doesn't exhibit this issue
* It seems that if we remove the first line ` VDP_setPlanSize(128, 32);`, it works fine.
* If we remove the sprite-related calls, then it renders fine *except* it will still render a small tile  wrongly (
[image]

timotei, 2017-12-22:
@Stephane-D I looked at the changes between 1.31 and 1.32, and stumbled over: https://github.com/Stephane-D/SGDK/compare/v1.31...v1.32#diff-1acaaa419798daa758b5ca7a5dd38ecf

It seems that the default sizes were changed, so maybe the SGDK considers a smaller amount of memory as usable by the user? At the beginning it was 4096 bytes (64x64 or 128x32), but now it's 64x32 (2048). According to the specs, 128x32 is a valid plane size.

timotei, 2017-12-22:
@Stephane-D I just did a local build of SGDK with the `vdp.c` change reverted, and it works fine :)

Stephane-D, 2017-12-26:
Hi timotei,

In fact this is not a bug : since SGDK 1.32 i changed VRAM layout so more tiles are available to user but for that, i needed to reduce plan size to 64x32 (which requires only 2048 bytes). If you want to use 64x64 or 128x32 plan size, you need to re arrange VRAM layout (more specifically plan B and plan A location) using the VDP_setAPlanAddress(..), VDP_setBPlanAddress(..), ... methods.

closed 2017-12-26 by Stephane-D

timotei, 2017-12-26:
@Stephane-D Well, then it would be very helpful to have something mentioning this in the changelog, besides just the "* VDP: - tweaked default VRAM memory layout" line :), since it's pretty unexpected. Otherwise people would think they have a bug in their code and start to investigate (like I did)

I am not sure how changing the address fixes the solution? Looking at the VDP layout, they don't overwrite anything 🤔 Is there any doc on how to figure this out properly? Looking at my backgrounds, I actually use just 86x32, so maybe I can tweak it, to be less than 128x32, right?

Stephane-D, 2017-12-27:
You can use the same address for both plan but then you need to carefully set the scrolling values as plan will share the same tilemap. But i do understand the issue people meet because of that... I think i put a warning in the changelog when i did the layout change (it was some versions ago), still i plan to add an automatic VRAM layout so changing plan size will adjust VRAM layout automatically to avoid any overlap in tilemap. Then if user manually set plan address the VRAM layout will get back to manual allocation.
That may be a _acceptable_ solution ;)

timotei, 2017-12-27:
In order to get back to the previous behaviour in v1.31, I tried setting the spritelist address to the original value, as present in that commit: [https://github.com/Stephane-D/SGDK/commit/976ee82d26ccb77d51b61aebe8ef37428e6a2271#diff-1acaaa419798daa758b5ca7a5dd38ecf](https://github.com/Stephane-D/SGDK/commit/976ee82d26ccb77d51b61aebe8ef37428e6a2271#diff-1acaaa419798daa758b5ca7a5dd38ecf)

```
VDP_setAPlanAddress(0xE000);
VDP_setBPlanAddress(0xC000);
VDP_setSpriteListAddress(0xBC00);
```

This partially fixes the issue, but I still have some weird tiles on Plane B 🤔 (which are not happening on v1.31) Am I missing something?

heitorgcosta, 2017-12-27:
There are still the HScroll address, through `VDP_setHScrollTableAddress` and the and the SpriteList address through `VDP_setSpriteListAddress`. Setting all these values to the old values fixed the issue for me.

timotei, 2017-12-28:
@heitorgcosta That was it! The HScroll/WindowPlan were missing.

With the following it renders fine:

```
    VDP_setPlanSize(128, 32);
    VDP_setAPlanAddress(0xE000);
    VDP_setBPlanAddress(0xC000);
    VDP_setWindowPlanAddress(0xB000);
    VDP_setSpriteListAddress(0xBC00);
    VDP_setHScrollTableAddress(0xB800);
```

@Stephane-D I really think we should put this in the changelog or something, so it's clear what has changed and how to migrate (maybe have the "migration.txt" file). The changelog only mentions "vdp layout change" but it's not obvious from start that it has a real effect on user's code.

Stephane-D, 2018-01-02:
I will add a note for it, for users experiencing issues when upgrading from old SGDK.
Sorry for the troubles again !