Stephane-D/SGDK #105: Separate the library and the toolchain

LukeUsher, 2018-01-02:
In order to allow easier updates when using an alternative toolchain, for example, using Linux with the gendev script, I think it would be much more convenient if the SGDK libraries could be in their own repository, separate from the precompiled toolchain binaries. This way it would be possible to integrate SGDK as a sub-module in another repository and keep it synced with upstream.

The current behavior of including windows binaries is nice for Windows users, and could be preserved by providing an SGDK-windows repository that brings in the parent SGDK repository as a submodule, alongside the precompiled tools

minerscale, 2018-01-02:
As a user of gendev, I very much agree with this.

Sentient68k, 2018-12-25:
Also a Gendev user here and I think this would be a good idea too.

doragasu, 2020-04-19:
I agree. Even when using Windows, sometimes you might not want to use the compiler included in SGDK (I use other with newlib support).

doragasu, 2021-12-19:
I am willing like to work in this direction and submit pull-requests, but I do not know if @Stephane-D would be interested in merging these kind of changes. Any words about it?

jagiella, 2022-06-22:
I am developing on ubuntu (22.10) and use the default m68k compiler/toolchain from the ubuntu repository (currently version 11.3.0):
`sudo apt install gcc-m68k-linux-gnu g++-m68k-linux-gnu`

For the build I installed the package of [gendev](https://github.com/kubilus1/gendev) bundling SGDK and another (in my opinion outdated) m68k toolchain (omitting in favor of ubuntu one). 

The SGDK provided Makefile for building roms just needs to change the toolchain part:
```Makefile
BIN := /usr/bin/m68k-linux-gnu-

CC := $(BIN)gcc
CPP := $(BIN)g++
LD := $(BIN)ld
NM := $(BIN)nm
OBJCPY := $(BIN)objcopy
```