Stephane-D/SGDK #106: undefined reference to SPR_setDepthSorting

werton, 2018-01-22:
I can not compile the starfield_donuts example, the compilation is interrupted
on line 87:
`SPR_setDepthSorting(spr, TRUE);`
with an error:
`main.c|87|undefined reference to SPR_setDepthSorting`
Without this line the example works, although not correctly, but i could not even find SPR_setDepthSorting method in the help, it seems that it does not exist.

Stephane-D, 2018-01-22:
Oh indeed i forgot to update this sample, removing the line is enough to make it work though.
Why are you thinking the example is not working correctly ?

werton, 2018-01-22:
I thought that these artifacts were due to the disabling of z sorting. Tried on three different emulators. Is this the way it should be?
[image]
[image]

Stephane-D, 2018-01-22:
Those artifacts are because of the sprite limit per scanline, i should tweak the donuts arrangement to avoid it ;)
Thanks for your report, i fixed the source file to remove the deprecated code :)

closed 2018-01-22 by Stephane-D

werton, 2018-01-22:
Oh, yes indeed, i did not think about it, thank you :)
Stephane-D, excuse me for offtop, i know that this is not the place for this, i just do not know where else to turn. i will delete this, as you will answer (or do not answer), but could you help me please? I want to register on Spritesmind's forum, but there is disabled registration due to spam. I'm sent email to administrator, but he does not respond. Could you to write to him that there are who wish to register?
ps: sorry for my english:(

Stephane-D, 2018-01-22:
Ok, but there is luck he don't see my message as well :-/ Can you tell me which nickname / email you used on the forum to subscribe ?

werton, 2018-01-22:
Many thanks. Usually i sign up under the nickname: werton with email: weerbox@gmail.com. I wrote to the administrator from the same mail.

Stephane-D, 2018-01-22:
ok, i just sent him a message about it =)

werton, 2018-01-22:
Thank you very much! :) Can I delete these messages that are not related to SGDK?

Stephane-D, 2018-01-22:
i'll do it ;)