Stephane-D/SGDK #108: lz4w does nothing for file names/paths that contain a capital letter

andwn, 2018-02-14:
I was wondering why it worked when rescomp called it and not when I did from a Makefile. Turns out that's because "pack.in" is all lowercase. This only effects OS/filesystems with case sensitive names, probably.
```
$ java -jar ~/mars/bin/lz4w.jar p res/Stage/Almond.pxm test.cpxm
res/stage/almond.pxm
$ mv res/Stage res/stage
$ mv res/stage/Almond.pxm res/stage/almond.pxm
$ java -jar ~/mars/bin/lz4w.jar p res/Stage/Almond.pxm test.cpxm
Packing res/stage/almond.pxm...
Stats:
  Num segment = 160 - 30 long match(s) - 0 miss(es)
  Mean literal size = 1.6
  Mean match size = 6.0
Initial size 2408 --> packed to 882 (36%)
```

Stephane-D, 2018-02-28:
oh indeed, stupid bug i wasn't aware because of windows OS doesn't take care of it. Thanks for reporting, will be fixed soon !

Stephane-D, 2018-03-13:
fixed in WIP

closed 2018-03-13 by Stephane-D