Stephane-D/SGDK #109: When 2 different notes are placed immediately before and after a loop, XGM continues playing the note/instrument from the end of the song

andwn, 2018-02-15:
[Minimal Example](https://github.com/Stephane-D/SGDK/files/1726288/test.zip)

In the attached DMF, at the start of the loop:
`A-2 7A 0`

And at the end of the song, on the same line as the loop marker (`0B 01`):
`C#5 -- 0`

After the output VGM is converted to XGM and played in game, the loop will play that first line (`A-2`) and many consecutive ones as `C#5` instead.

This also effects instruments. If for example the `C#5` line used instrument 5 the `A-2` line would also be played with instrument 5 instead of 0 like it should.

While reproducing with the test track I noticed this does not happen if the destination of the loop is the very start (`0B 00`).

Stephane-D, 2018-02-28:
Thanks for providing a minimal reproducible test case ! I need to investigate in that, i guess it's a problem from the VGM to XGM converter.

clbr, 2018-04-04:
Probably the same bug as reported here:
http://gendev.spritesmind.net/forum/viewtopic.php?f=19&t=2618

Stephane-D, 2019-08-19:
Should be fixed (finally ^^)

closed 2019-08-19 by Stephane-D

andwn, 2019-08-20:
Awesome! I'll have to upgrade when I have some free time