Stephane-D/SGDK #113: how to see the equivalent of some functions that I believe in my source code in 68k?

loboere, 2018-03-14:
example  

```
u16 sum(u16 n)
{
return n+n
}
```

como ver esta funcion en 68k?
puedo exportar en un 68k_asm.txt?
puedo exportar todo el codigo fuente en 68k_ams.txt?

Stephane-D, 2018-03-14:
Just use the 'asm' target of makefile to produce assembly listing : 
`make -f makefile.gen asm`

You should find .lst files in your /out folder of your project containing generated assembly code :)

closed 2018-03-14 by Stephane-D

loboere, 2018-03-14:
I get this error
`make: *** No rule to make target `asm'.  Stop. `

Stephane-D, 2018-03-14:
Oh yeah, the new makefile is not yet in the SGDK archive, you should grab it from the repository :  
https://github.com/Stephane-D/SGDK/blob/master/makefile.gen

sorry for the trouble.