Stephane-D/SGDK #117: XGM stop/start issue

matteusbeus, 2018-04-05:
If you put a VDP_waitVSync straight after a XGM_stopPlay you crash the XGM driver.  XGM_startPlay refuses to restart.

`void playMusic(const u8 *song) {
    stopMusic();
    XGM_startPlay(song);
}

void stopMusic() {
    XGM_stopPlay();
    VDP_waitVSync();
}
`

matteusbeus, 2018-12-27:
Any news on this? I'm getting close to a full release and it would be nice to know a work around :)

Stephane-D, 2018-12-27:
In fact the XGM stop / start are quite timing sensitive, can you use XGM_pausePlay() / XGM_startplay() command instead ? In general that is enough to fix the problem :)

matteusbeus, 2018-12-27:
Thanks! I'll give it a go and see what happens :D

Stephane-D, 2018-12-27:
Also sometime for stop command, just using 2 VDP_waitVSync instead of 1 is enough to fix the timing issue (because Z80 interpret 68k commands once per frame, but timing isn't exactly a frame, it can be a bit less or more so it's better to wait 2 frames between each command)

matteusbeus, 2018-12-27:
When you say 2 vsync you mean 1 before and 1 after the stop?

Stephane-D, 2018-12-27:
Better to put 2 after to let time for the command to be executed.

matteusbeus, 2018-12-27:
Unfortunately it results in complete silence and the driver won't restart.

Stephane-D, 2018-12-27:
That is really weird, did you try to use pause instead of stop command ?

matteusbeus, 2018-12-27:
No I'll try that a little later and post back here.

Stephane-D, 2018-12-27:
ok :)

moon-watcher, 2018-12-27:
What I coded for AbbayeMD https://github.com/moon-watcher/AbbayeMD/blob/master/game/src/music.c

Stephane-D, 2018-12-29:
Oh so you needed 3 waitVSync, that seems a lot..

Stephane-D, 2021-07-08:
XGM driver got some small fixes lately (take the last sources from repository). The different issues with stop/start commands not always properly working should be fixed. I also recommend to not use XGM_stop() command but XGM_pausePlay() instead as XGM_stop() may need 2 or 3 frames to apply when XGM_pause should take a bit more than 1 frame at worst. Also thanks to Matt from BigEvilCorp who pointed a bug which could let a small remaining sound even after the pause command (DAC write wasn't properly restored) so now you can use it safely.

closed 2021-07-08 by Stephane-D