Stephane-D/SGDK #122: A Question About SGDK.

mouhamed1995, 2018-04-09:
Hello.
Can You Please Tell As About 
This Function:
VDP_fillTileMapRect(....................................................);
Thanx.

closed 2018-04-09 by Stephane-D

mouhamed1995, 2018-04-09:
Pourquoi Tu a fermé mon topic ?????

Stephane-D, 2018-04-10:
Parce-que la manière dont le message était rédigé fait penser à un spam / bot.
La question n'a pas de sens telle qu'elle est posée, en plus ce n'est pas l'endroit pour poser ce genre de question..

mouhamed1995, 2018-04-10:
je excuse pour la deronegement.
est' ce que il ya un forum tu as un groupe en lui pour poser cette question
????
merci.

mouhamed1995, 2018-05-24:
> 2018-04-10 7:15 GMT-07:00 Stephane Dallongeville <notifications@github.com
> Voici un forum où vous pourrez trouver des réponses concernant SGDK et
> son utilisation :

salut.
je veux pour questioner un autre question.
et ce que le sgdk marche a visual studio 6.

Stephane-D, 2018-05-24:
Salut,

Encore une fois, utilise le forum prévu pour ce genre de question : 
http://gendev.spritesmind.net/forum/viewforum.php?f=19

Tu verras que certaines questions ont déjà été posées par rapport à Visual Studio :
http://gendev.spritesmind.net/forum/viewtopic.php?f=19&t=2564

Pour Visual Studio 6, je dirais que si tu as la possibilité de régler le compilateur alors ça doit être possible, mais Visual Studio 6 c'est très vieux ! Donc je ne suis pas sur que cela soit possible.