Stephane-D/SGDK #125: fix32Mul?

timotei, 2018-05-19:
Hello,

I just did the following sample test which seems it overflows? I am trying to do some calculations that involve doing the squared of a floating-point number:

```
 fix32 x = intToFix32(200);
    fix32 xsquared = fix32Mul(x, x);
    KLog_F2("x: ", x, ", squared: ", xsquared);
```

This prints:

`Message : x: 200.00, squared: -960.00`

Am I missing something?

Stephane-D, 2018-05-22:
That's indeed a overflow issue.
There is several strategy to adopt when doing fixed point multiplication or division, unfortunately each strategy has its drawbacks... here we loss accuracy with high value numbers. I can change the strategy but then we will loss accuracy on small number. Still i may change for something intermediate (but operation will be slower then).
What you can do is to "pre shift" your values first:

>  fix32 x = intToFix32(200);
>  x >>= 5;
>  fix32 xsquared = fix32Mul(x, x) << 10;

Stephane-D, 2018-10-09:
Can't be fixed (except by using very slow 64 bit multiplication / division but not what we want here)

closed 2018-10-09 by Stephane-D