Stephane-D/SGDK #126: Error in when using -03 optimization

moon-watcher, 2018-06-10:
I've been using this linked link code https://github.com/moon-watcher/GrielsQuest/blob/master/game/libs/vram.c in my previous projects. 

Due to a -03 optimization, current SGDK throws an error [image] in line 87  https://github.com/moon-watcher/GrielsQuest/blob/d7da36ac310aa2f6654588bf4ea891b86ceaebc7/game/libs/vram.c#L87
`node->next = new;`

I changed to -01 and the rom worked again

Stephane-D, 2018-06-11:
To be honest i don't see immediately what is wrong and i'm almost certain this is not a "compiler bug". Still be careful if you use the memory manager (MEM_xxx calls), be sure that you don't mix calls from VInt callback and "main loop" in which case you can corrupt its state. Also maybe you don't know but there is a VRAM memory manager in SGDK ;)
https://github.com/Stephane-D/SGDK/blob/master/inc/vram.h
You create a VRAM region (with a specified size) then you just allocate out of it :)

moon-watcher, 2018-07-28:
Found it! -03 optimization it NEEDS this:
```
if (node)
    node->next = new;
```

closed 2018-07-28 by moon-watcher

Stephane-D, 2018-07-31:
Oh indeed, i would even say you need it in all case and not only in -O3 as node can be null at this point. You are even doing that at the beginning :  
```
if ( !node )
{
		_list = new;
}
```

And you don't update node so it remains null here.. not sure that is what you want.
Also you should check for null for the whole piece of code (ans not only for node->next affectation) as it doesn't make sense to execute it if you don't use the 'new' block.