Stephane-D/SGDK #127: Feature request: 32x support or sample code

ryanyoder, 2018-06-10:
My goal is to build a 32x game and your SGDK is a great start with the Genesis side of the equation.
Can you provide sample code on initializing the 32x processing as part of a SGDK app and then even better, build some simple libraries to do things such as rendering to the video buffers and flipping them etc?

If not, I might end up providing this sample code when I figure it out.

matteusbeus, 2018-06-10:
That's a huge ask!

ryanyoder, 2018-06-10:
It might not be that simple, that's true.  I'll look into it and when I get something basic working I'll post it here.

Stephane-D, 2018-06-11:
32X dev requires much more than what SGDK currently has (SHx compiler and binutils) and is currently out of scope for me.
Chilly Willy developed a minimalist 32X dev environment, i think you can find it digging in this 32X dedicated forum section :
http://gendev.spritesmind.net/forum/viewforum.php?f=4

closed 2018-06-11 by Stephane-D