Stephane-D/SGDK #131: Un Question???

mouhamed1995, 2018-09-28:
Je Vais Proposer Un Question.
est ce que Vous-Avez Allez Publier Un Livre Parceque Un Idea Pour Publiez Un Livre Son Theme Est A La SGDK Est Un Idea Ne Pas Un Idiot Idea.
Merci.

Stephane-D, 2018-09-28:
Which sprite update management ? :p

Stephane-D, 2018-09-28:
> 
> 
> Je Vais Proposer Un Question.
> est ce que Vous-Avez Allez Publier Un Livre Parceque Un Idea Pour Publiez Un Livre Son Theme Est A La SGDK Est Un Idea Ne Pas Un Idiot Idea.
> Merci.

Of course no, i can't wrote a real book for that :-/ Ii do that as a hobby and have unfortunately very few time to dedicate to it, but i will improve the tutorials for sure =)

closed 2018-09-28 by Stephane-D

matteusbeus, 2018-09-28:
Happy to work with you on that Stef

matteusbeus, 2018-09-28:
The defrag or has that been rolled out to live now?

CristianoSword, 2018-09-28:
I think the idea of ​​making a book or e-book would be nice, but it would take a lot of the person to do it, besides time would have to have a good knowledge of SGDK, I would find more interesting simple tutorials on how to make a platform game Mario type 1 for example.

Stephane-D, 2018-10-01:
SGDK is just to develop on Sega Megadrive, doing a book about making game is a complete different story ;)