Stephane-D/SGDK #133: Varargs warning on sprintf

jackbonaguro, 2018-10-23:
On Windows 10 64-bit

make a new project with only one file - main.c:
```
#include <genesis.h>

int main() {
	char * text = "";
	sprintf(text, "Hello World");
	return 0;
}

/*
Now run:
	%GDK_WIN%/bin/make -f %GDK_WIN%/makefile.gen
*/
```

And run the command `%GDK_WIN%/bin/make -f %GDK_WIN%/makefile.gen` from the project root.

I am getting the following warning:

```
C:/sgdk/src/string.c: In function 'sprintf.constprop':
C:/sgdk/src/string.c:648:5: warning: second parameter of 'va_start' not last named argument [-Wvarargs]
     va_start(args, fmt);
     ^
```

I'm pretty sure this warning is incorrect.

jackbonaguro, 2018-10-23:
Seems like a compiler issue, but I don't know what version of gcc is used.

matteusbeus, 2018-10-25:
I've had this problem myself in the past! usually due to having the wrong SGDK version installed or not rebuilding the library after an SGDK version upgrade.

jaytagdamian, 2020-06-04:
I have exactly the same issue with SGDK 1.51, even when having no other gcc version installed on my workstation.

If I comment out this line of code:
sprintf(str_player_score,"%d",player_score);

The error disappears.

Stephane-D, 2020-06-04:
Too bad, i was optimist about the last changes to resolve the issues.
Did you rebuild the library in between as i didn't pushed last lib binaries after the change.

jaytagdamian, 2020-06-09:
I had previously downloaded https://github.com/Stephane-D/SGDK/releases/download/v1.51/sgdk151.7z and followed the instructions at: https://github.com/Stephane-D/SGDK/wiki/SGDK-Installation

This time, to ensure I was using the latest configuration and libraries, I cloned the repository, and then ran build_lib.bat.

Now, when I d:/sgdk/bin/make -f d:\sgdk\makefile.gen the error is gone.

As a test, I tried running build_lib.bat from the existing directory, but the error persists.

Solution:
Download https://github.com/Stephane-D/SGDK/archive/master.zip
Overwrite the existing sgdk install directory with the contents
Run build_lib.bat

Stephane-D, 2020-06-09:
Yeah the fix is only available through direct repository content (not in SGDK 1.51) and it really requires rebuilding the library, still i don't understand why the build_lib.bat batch didn't made it !
At least it seems the bug is resolved :)

closed 2020-06-09 by Stephane-D