Stephane-D/SGDK #136: "Hello World!" gives error

ID453543435, 2018-12-09:
#include <genesis.h>

int main()
{
	VDP_drawText("Hello World!", 10, 13);
	return (0);
}
gives error
https://i.ibb.co/JxcCtS3/image.png

https://github.com/Stephane-D/SGDK/wiki/Tuto-Hello-World

ehaliewicz, 2018-12-09:
Try adding an empty while loop after the VDP_drawText call.  You cannot return from the main function on the genesis.
```#include <genesis.h>

int main()
{
VDP_drawText("Hello World!", 10, 13);
while(1) { }
return (0);
}

ID453543435, 2018-12-09:
do update the 

https://github.com/Stephane-D/SGDK/wiki/Tuto-Hello-World

page

Stephane-D, 2018-12-10:
Did you carefully read this tutorial ? :p 
It's explained that your can't use that code on Sega Gensis and add the infinite loop:

> This code is correct but not Genesis friendly : do you think we can return (and stop) a program ?
> 
> Video games repeatedly update the TV screen, and it's up to you to handle things before or while a refresh.
> 
> So a more Genny Hello World is more like this :
> 
> #include <genesis.h>
> 
```
 int main()
 {
 	VDP_drawText("Hello Genny World!", 10, 13);
 
 	while(1)
 	{
 		//read input
 		//move sprite
 		//update score
 		//draw current screen (logo, start screen, settings, game, gameover, credits...)
 	}
 	return (0);
 }
```
> 
> This code is correct, even if return(0) will never be reached.

closed 2018-12-10 by Stephane-D

ID453543435, 2019-01-01:
no one read english langiage code
the C code incorrect