Stephane-D/SGDK #140: memmove implementation

lab313ru, 2019-03-25:
It would be great to add this function.

polluks, 2020-09-03:
vbcc's implementation
````
void *memcpy(__reg("a0") void *, __reg("a1") const void *, __reg("d2") size_t) =
        "\tinline\n"
        "\tmove.l\ta0,d0\n"
        "\tsubq.l\t#4,d2\n"
        "\tbcs\t.l3\n"
        "\tmove.l\td0,d1\n"
        "\tlsr.l\t#1,d1\n"
        "\tbcc\t.l1\n"
        "\tmove.b\t(a1)+,(a0)+\n"
        "\tsubq.l\t#1,d2\n"
        "\tbcs\t.l3\n"
        "\taddq.l\t#1,d1\n"
        ".l1\n"
        "\tlsr.l\t#1,d1\n"
        "\tbcc\t.l2\n"
        "\tmove.w\t(a1)+,(a0)+\n"
        "\tsubq.l\t#2,d2\n"
        "\tbcs\t.l3\n"
        ".l2\n"
        "\tmove.l\t(a1)+,(a0)+\n"
        "\tsubq.l\t#4,d2\n"
        "\tbcc\t.l2\n"
        ".l3\n"
        "\taddq.l\t#3,d2\n"
        "\tbcc\t.l5\n"
        ".l4\n"
        "\tmove.b\t(a1)+,(a0)+\n"
        "\tdbf\td2,.l4\n"
        ".l5\n"
        "\teinline";
void *__memcpy_desc(__reg("a0") void *, __reg("a1") const void *,
                    __reg("d2") size_t) =
        "\tinline\n"
        "\tsubq.l\t#4,d2\n"
        "\tbcs\t.l3\n"
        "\tmove.l\ta0,d0\n"
        "\tlsr.l\t#1,d0\n"
        "\tbcc\t.l1\n"
        "\tmove.b\t-(a1),-(a0)\n"
        "\tsubq.l\t#1,d2\n"
        "\tbcs\t.l3\n"
        ".l1\n"
        "\tlsr.l\t#1,d0\n"
        "\tbcc\t.l2\n"
        "\tmove.w\t-(a1),-(a0)\n"
        "\tsubq.l\t#2,d2\n"
        "\tbcs\t.l3\n"
        ".l2\n"
        "\tmove.l\t-(a1),-(a0)\n"
        "\tsubq.l\t#4,d2\n"
        "\tbcc\t.l2\n"
        ".l3\n"
        "\taddq.l\t#3,d2\n"
        "\tbcc\t.l5\n"
        ".l4\n"
        "\tmove.b\t-(a1),-(a0)\n"
        "\tdbf\td2,.l4\n"
        ".l5\n"
        "\tmove.l\ta0,d0\n"
        "\teinline";
#define memmove(d,s,n) ((d)<=(s) ? memcpy(d,s,n) : \
                                   __memcpy_desc((char *)(d)+(n),(char *)(s)+(n),n))
````

polluks, 2020-09-03:
vbcc's implementation
````
void *memcpy(__reg("a0") void *, __reg("a1") const void *,
             __reg("d2") size_t) =
        "\tinline\n"
        "\tmove.l\ta0,d0\n"
        "\tcmp.l\t#16,d2\n"
        "\tblo\t.l5\n"
        "\tmoveq\t#1,d1\n"
        "\tand.b\td0,d1\n"
        "\tbeq\t.l1\n"
        "\tmove.b\t(a1)+,(a0)+\n"
        "\tsubq.l\t#1,d2\n"
        ".l1\n"
        "\tmove.l\ta1,d1\n"
        "\tand.b\t#1,d1\n"
        "\tbeq\t.l3\n"
        "\tcmp.l\t#$10000,d2\n"
        "\tblo\t.l5\n"
        ".l2\n"
        "\tmove.b\t(a1)+,(a0)+\n"
        "\tsubq.l\t#1,d2\n"
        "\tbne\t.l2\n"
        "\tbra\t.l7\n"
        ".l3\n"
        "\tmoveq\t#3,d1\n"
        "\tand.l\td2,d1\n"
        "\tsub.l\td1,d2\n"
        ".l4\n"
        "\tmove.l\t(a1)+,(a0)+\n"
        "\tsubq.l\t#4,d2\n"
        "\tbne\t.l4\n"
        "\tmove.w\td1,d2\n"
        ".l5\n"
        "\tsubq.w\t#1,d2\n"
        "\tblo\t.l7\n"
        ".l6\n"
        "\tmove.b\t(a1)+,(a0)+\n"
        "\tdbf\td2,.l6\n"
        ".l7\n"
        "\teinline";
void *__memcpy_desc(__reg("a0") void *, __reg("a1") const void *,
                    __reg("d2") size_t) =
        "\tinline\n"
        "\tcmp.l\t#16,d2\n"
        "\tblo\t.l5\n"
        "\tmoveq\t#1,d1\n"
        "\tmove.l\ta0,d0\n"
        "\tand.b\td1,d0\n"
        "\tbeq\t.l1\n"
        "\tmove.b\t-(a1),-(a0)\n"
        "\tsubq.l\t#1,d2\n"
        ".l1\n"
        "\tmove.l\ta1,d0\n"
        "\tand.b\td1,d0\n"
        "\tbeq\t.l3\n"
        "\tcmp.l\t#$10000,d2\n"
        "\tblo\t.l5\n"
        ".l2\n"
        "\tmove.b\t-(a1),-(a0)\n"
        "\tsubq.l\t#1,d2\n"
        "\tbne\t.l2\n"
        "\tbra\t.l7\n"
        ".l3\n"
        "\tmoveq\t#3,d1\n"
        "\tand.l\td2,d1\n"
        "\tsub.l\td1,d2\n"
        ".l4\n"
        "\tmove.l\t-(a1),-(a0)\n"
        "\tsubq.l\t#4,d2\n"
        "\tbne\t.l4\n"
        "\tmove.w\td1,d2\n"
        ".l5\n"
        "\tsubq.w\t#1,d2\n"
        "\tblo\t.l7\n"
        ".l6\n"
        "\tmove.b\t-(a1),-(a0)\n"
        "\tdbf\td2,.l6\n"
        ".l7\n"
        "\tmove.l\ta0,d0\n"
        "\teinline";
void *memset(__reg("a0") void *, __reg("d0") int, __reg("d2") size_t) =
        "\tinline\n"
        "\tmove.l\ta0,a1\n"
        "\tcmp.l\t#16,d2\n"
        "\tblo\t.l3\n"
        "\tmove.l\ta0,d1\n"
        "\tand.b\t#1,d1\n"
        "\tbeq\t.l1\n"
        "\tmove.b\td0,(a0)+\n"
        "\tsubq.l\t#1,d2\n"
        ".l1\n"
        "\tmove.b\td0,d1\n"
        "\tlsl.w\t#8,d0\n"
        "\tmove.b\td1,d0\n"
        "\tmove.w\td0,d1\n"
        "\tswap\td0\n"
        "\tmove.w\td1,d0\n"
        "\tmoveq\t#3,d1\n"
        "\tand.l\td2,d1\n"
        "\tsub.l\td1,d2\n"
        ".l2\n"
        "\tmove.l\td0,(a0)+\n"
        "\tsubq.l\t#4,d2\n"
        "\tbne\t.l2\n"
        "\tmove.w\td1,d2\n"
        ".l3\n"
        "\tsubq.w\t#1,d2\n"
        "\tblo\t.l5\n"
        ".l4\n"
        "\tmove.b\td0,(a0)+\n"
        "\tdbf\td2,.l4\n"
        ".l5\n"
        "\tmove.l\ta1,d0\n"
        "\teinline";

#define memmove(d,s,n) ((d)<=(s) ? memcpy(d,s,n) : \
                                   __memcpy_desc((char *)(d)+(n),(char *)(s)+(n),n))
````

polluks, 2020-09-03:
vbcc's implementation
````
void *memcpy(__reg("a0") void *, __reg("a1") const void *,
             __reg("d2") size_t) =
        "\tinline\n"
        "\tmove.l\ta0,d0\n"
        "\tcmp.l\t#16,d2\n"
        "\tblo\t.l5\n"
        "\tmoveq\t#1,d1\n"
        "\tand.b\td0,d1\n"
        "\tbeq\t.l1\n"
        "\tmove.b\t(a1)+,(a0)+\n"
        "\tsubq.l\t#1,d2\n"
        ".l1\n"
        "\tmove.l\ta1,d1\n"
        "\tand.b\t#1,d1\n"
        "\tbeq\t.l3\n"
        "\tcmp.l\t#$10000,d2\n"
        "\tblo\t.l5\n"
        ".l2\n"
        "\tmove.b\t(a1)+,(a0)+\n"
        "\tsubq.l\t#1,d2\n"
        "\tbne\t.l2\n"
        "\tbra\t.l7\n"
        ".l3\n"
        "\tmoveq\t#3,d1\n"
        "\tand.l\td2,d1\n"
        "\tsub.l\td1,d2\n"
        ".l4\n"
        "\tmove.l\t(a1)+,(a0)+\n"
        "\tsubq.l\t#4,d2\n"
        "\tbne\t.l4\n"
        "\tmove.w\td1,d2\n"
        ".l5\n"
        "\tsubq.w\t#1,d2\n"
        "\tblo\t.l7\n"
        ".l6\n"
        "\tmove.b\t(a1)+,(a0)+\n"
        "\tdbf\td2,.l6\n"
        ".l7\n"
        "\teinline";
void *__memcpy_desc(__reg("a0") void *, __reg("a1") const void *,
                    __reg("d2") size_t) =
        "\tinline\n"
        "\tcmp.l\t#16,d2\n"
        "\tblo\t.l5\n"
        "\tmoveq\t#1,d1\n"
        "\tmove.l\ta0,d0\n"
        "\tand.b\td1,d0\n"
        "\tbeq\t.l1\n"
        "\tmove.b\t-(a1),-(a0)\n"
        "\tsubq.l\t#1,d2\n"
        ".l1\n"
        "\tmove.l\ta1,d0\n"
        "\tand.b\td1,d0\n"
        "\tbeq\t.l3\n"
        "\tcmp.l\t#$10000,d2\n"
        "\tblo\t.l5\n"
        ".l2\n"
        "\tmove.b\t-(a1),-(a0)\n"
        "\tsubq.l\t#1,d2\n"
        "\tbne\t.l2\n"
        "\tbra\t.l7\n"
        ".l3\n"
        "\tmoveq\t#3,d1\n"
        "\tand.l\td2,d1\n"
        "\tsub.l\td1,d2\n"
        ".l4\n"
        "\tmove.l\t-(a1),-(a0)\n"
        "\tsubq.l\t#4,d2\n"
        "\tbne\t.l4\n"
        "\tmove.w\td1,d2\n"
        ".l5\n"
        "\tsubq.w\t#1,d2\n"
        "\tblo\t.l7\n"
        ".l6\n"
        "\tmove.b\t-(a1),-(a0)\n"
        "\tdbf\td2,.l6\n"
        ".l7\n"
        "\tmove.l\ta0,d0\n"
        "\teinline";

#define memmove(d,s,n) ((d)<=(s) ? memcpy(d,s,n) : \
                                   __memcpy_desc((char *)(d)+(n),(char *)(s)+(n),n))
````