Stephane-D/SGDK #141: Missing { and failed to open "libres.h"

VreodiumLaurence, 2019-03-29:
So I wrote a basic code
`#include <genesis.h>

int main()
{
	while (1)
	{
		VDP_drawText("Hello World! Greetings in Sega Genesis!", 10, 13);
		VDP_waitVSync();
	}

	return(0);
}`

All is fine and dandy but then I got these errors:
font.h - cannot open source file "libres.h" Line 18
string.h - expected a '{' Line 286
What is going on, what is missing?

Stephane-D, 2019-04-01:
Check that your SGDK installation is correct, also for the libres.h, you have it in the "release" packages.
But you can also regenerate it by re-building the library :)

closed 2019-04-01 by Stephane-D

amaurysilva, 2020-04-12:
I have the same problem.
I downloaded the last GDK but the problem continues.
Please help me!! :-(

trialforce, 2020-11-27:
My project were working fine, then the same error starts from nothing.