Stephane-D/SGDK #146: Windows installation/build issues

leftnoodle, 2019-10-15:
Hi Stef,

Just downloaded the source today and setting it up on a new laptop. 

There are some build issues:
C:SGDK/res/libres.h:1:0: error: unterminated #ifndef

After fixing that there are then some other errors all referencing font_default:

C:SGDK/src/vdp.c: In function 'VDP_init':
C:SGDK/src/vdp.c:129:24: error: 'font_default' undeclared (first use in this function)
     if (!VDP_loadFont(&font_default, 0))

C:SGDK/src/vdp.c: In function 'updateMapsAddress':
C:SGDK/src/vdp.c:844:23: error: 'font_default' undeclared (first use in this function)
         VDP_loadFont(&font_default, TRUE);

I haven't fixed these yet as I am currently at work but thought I should report the issue before I forgot!

Kind regards,

Simon

leftnoodle, 2019-10-16:
Ok this was even stranger, downloaded a fresh zip today and everything is fine, the other zip appears to be corrupted. Apologies.

closed 2019-10-16 by leftnoodle

Stephane-D, 2019-10-16:
I was really surprised with your report as everything was compiling fine here.. yeah maybe you got a corrupted zip for some reasons :-/