Stephane-D/SGDK #148: compiling error windows

molo32, 2020-01-02:
C:\SGDK>make.exe -f C:\SGDK\makefile.gen
C:\SGDK/bin/mkdir -p src/boot
C:\SGDK/bin/mkdir -p out
C:\SGDK/bin/mkdir -p out/src
C:\SGDK/bin/mkdir -p out/res
C:\SGDK/bin/ld -T C:\SGDK/md.ld -nostdlib --oformat binary -o out/rom_head.bin o
ut/rom_head.o
c:\SGDK\bin\ld.exe: cannot open linker script file C:SGDK/md.ld: No such file or
 directory
make.exe: *** [out/rom_head.bin] Error 1

The md.ld file is present in c: / sgdk
I also have correct environment variables
because he tells me that md.ld is not present?

andwn, 2020-01-02:
Use forward slashes (/) in your GDK environment variable. The backslash (\) is getting stripped, notice C:SGDK instead of C:\SGDK.

Stephane-D, 2020-01-03:
Indeed it seems to be an issues with your environment variables.
GDK should use '/' and GDK_WIN should use '\' character.

closed 2020-01-03 by Stephane-D