Stephane-D/SGDK #151: V1.41 linker error with memset

bizulk, 2020-02-28:
Hello !

First thank you for that amazing work.
I am starting with tutorial, install the stuff, recompiled library, and (re)built library.
Next I wanted a to code a key logger from the joypad tutorial but I'm stuck with a linker problem, here is the log : 

```
C:/Users/admin/Documents/Developpement/sgdk141/bin/gcc -m68000 -Wall -fno-builtin -Iinc -Isrc -Ires -IC:/Users/admin/Documents/Developpement/sgdk141/inc -IC:/Users/admin/Documents/Developpement/sgdk141/res -BC:/Users/admin/Documents/Developpement/sgdk141/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -c src/main.c -o out/src/main.o
C:/Users/admin/Documents/Developpement/sgdk141/bin/gcc -BC:/Users/admin/Documents/Developpement/sgdk141/bin -n -T C:/Users/admin/Documents/Developpement/sgdk141/md.ld -nostdlib out/sega.o @out/cmd_ C:/Users/admin/Documents/Developpement/sgdk141/lib/libmd.a C:/Users/admin/Documents/Developpement/sgdk141/lib/libgcc.a -o out/rom.out
`memset' referenced in section `.text' of C:\Users\admin\AppData\Local\Temp\ccuTQGW1.ltrans1.ltrans.o: defined in discarded section `.text' of memory.o (symbol from plugin
```

I tried re-implementing in my source file the memset function but it does not work (redundant définition).

With some googling I [found out](https://mcuoneclipse.com/2017/11/12/solving-problem-with-gnu-linker-and-referenced-in-section-defined-in-discarded-section-error-message/) that the "lto" optimization may cause this issue.

But I may be wrong, since that problem with memset would have been found with somebody else sooner...
[main.txt](https://github.com/Stephane-D/SGDK/files/4269357/main.txt)

I attach my test file.

bizulk, 2020-03-02:
I did recompile the library changing memset definition like this : 
`void __attribute__ ((noinline, used)) memset(void *to, u8 value, u16 len)`

and now compiling the project with release conf does work.

Stephane-D, 2020-03-04:
You're not the first one experiencing this issue.
Indeed with LTO optimization the compiler tend to often inline memset / memcpy methods and if you don't reference them directly in your code it can be completely removed from linking process, then the problem arrive when you're doing (big) structure initialization or affectation:
`char szMsg[100]  = {0};`
`myStruct = struct2;`
So this is the initialization which internally will be optimized to memset but linker removed it as it was not explicitly called.
Your fix actually works but i guess that using
`memset(szMsg, 0, sizeof(szMsg));` may also fix the problem as you explicitly call memset :)

bizulk, 2020-03-05:
Hello !

I did have a doubt that memset would be called implicitely but could not conclude to it... "Do the compiler really do that, I tought ?" 
Maybe with some listing assembly I would have seen it.

I think you're right to apply the patch as many people will not guess what to do with this error.

Stephane-D, 2020-03-05:
Done :)

closed 2020-03-05 by Stephane-D