Stephane-D/SGDK #153: About tile tutorial : trying to compile a 8bbp image

bizulk, 2020-03-10:
Hello !

 I have write a small image test from the tutorial. The tutorial reports the constraint : "
the bitmap should be 4bpp or 8bpp ..."

My test works with 4 bbp image, but not with 8 bbp image. I tried png and bitmap, but the compiler always return an error like : 

```
Resource: IMAGE MatthiasSelso "MatthiasSelso_gimp_200x240_8bbp_64color.png" 0
--> executing plugin IMAGE...
Error: pixel at [80,0] reference a different palette.
Error: cannot compile resource 'IMAGE MatthiasSelso "MatthiasSelso_gimp_200x240_8bbp_64color.png" 0'
```

I generated these images with Gimp or imamagick, but the result is same....

Here an commande example with imagemagick : 
magick file.png -dither FloydSteinberg -colors 64 file_out.png

See attached an image example.
[image]

matteusbeus, 2020-03-11:
The mega drive can't display standard images that are 64 colours.  Your image must be made up of 4 palettes of 16 colours each palette with a transparent colour at position 0.  This means the image is actually 61 colours.  Furthermore the image must have a palette applied to each 8x8 tile of the image.  Using magick in the way you have specified will simply create an image of 64 colours and fail to follow the requirements I have mentioned.

Stephane-D, 2020-03-11:
Exactly what @matteusbeus mentionned. Not a bug, just a hardware limitation about how work the Megadrive.

closed 2020-03-11 by Stephane-D

bizulk, 2020-03-11:
Hello. 
@matteusbeus  Thanks for the answer, but the tutorial  already talk about this part, and my question is **about the rescomp tool**.
Please read carefully, I am quoting the tutorial part giving constraint to the compiler.

I tought the compiler could take the full image and create the 4 pals, because is accepts 8bbp image. Actually it takes the 200x220 résolution, that's more than a tile isn't it ?
So does the rescomp creates the 4 palettes with it's color (nowing that one is used for transparency) ?
Or shall I manually cut the image into tile in order to use 64 colors globally ?
Or use another tool that help me do that ?

Thanks

bizulk, 2020-03-11:
Well, I think the answer is partly in the tutorial : 

> Rescomp and others tools made by the community (Genitile, B2T, GenRes, Mega-Happy-Sprite...) let you convert a **16 color bitmap** in Genny's tile and pal

So my bitmap may be 8 bbp but can't be more than 16 colors, right ?
If I want to display 61 colors I may cut the image in 4 parts ?
Is there a tool that can do that by calculating some colors dispersion ?

Stephane-D, 2020-03-12:
> Hello.
> @matteusbeus Thanks for the answer, but the tutorial already talk about this part, and my question is **about the rescomp tool**.
> Please read carefully, I am quoting the tutorial part giving [...]

Rescomp is indeed able to use more than 1 palette if your image is more than 64 colors *but* you have to respect the hardware color constraints which are :
- each tile (block of 8x8 pixels) can reference only 1 palette of 16 colors
- up to 4 palettes of 16 colors maximum
- color 0 of each palette is transparent and cannot be used

There is not graphics tools as far i know which are able to deal with the "each tile (block of 8x8 pixels) can reference only 1 palette of 16 colors" constraint so the classic way of doing is just to restrict the whole image to 16 colors or having 2 images of 16 colors and combine them (using both hardware planes) to show a 31 colors image.

bizulk, 2020-03-12:
Hello !
That's the answer I was expecting !
I suggest adding that note in the wiki, or may I do it ?

> Rescomp is indeed able to use
so it can create the palettes ?!
But it will lead to nothing because there little chance of it to have only one referenced palette per tile.

to display more color I'll try your suggestion and mine (and maybe discover why I'm wrong :D).

Stephane-D, 2020-03-12:
> Hello !
> That's the answer I was expecting !
> I suggest adding that note in the wiki, or may I do it ?
>
>
> to display more color I'll try your suggestion and mine (and maybe discover why I'm wrong :D).

To be honest it's more about knowing the graphics constraint of the system.
The problem is that if we start speaking about these specific graphics constraints here then we probably need to get further about graphics system of the Megadrive. And honestly that is a *big* topic.

It's why the SGDK README file contains this : 

> First, you need to know that SGDK uses C language (assembly is also possible) so it's highly recommended to be familiar with C programming before trying to develop with SGDK. Learning C language at same time than learning 'MegaDrive' programming is (imo) definitely too difficult and you will end nowhere.
> It's also important to understand how the Sega Megadrive works internally (specifically the video system) so i recommend you to read documents about the Sega Megadrive hardware and development.
> For that you can visit Mega Cat Studios Sega Megadrive graphics guide page:
> https://megacatstudios.com/blogs/press/sega-genesis-mega-drive-vdp-graphics-guide-v1-2a-03-14-17
> You can also check Sik's web site which is more dedicated to assembly programming but explain a lot about the Sega Megadrive hardware:
> https://plutiedev.com/
> 
> Then when you feel ready you can go further and finally install SGDK :)

Again, it's ***really*** important to understand how the Megadrive hardware work before trying to use SGDK. I eventually may add more information about the Sega Megadrive hardware later but that is a big topic and i think some other dedicated websites / references will do it better than me ;)

bizulk, 2020-03-12:
OK.
I was wondering how much rescomp is to able to handle these constraints for me, but I'll keep in mind that its limited to **16 color bitmap**, if I want more I **have** to go deeper in the Megadrive hardware knowledge with the use of palette, tiles, planes presented in the tuto.
That's outside the tutorial target.

Thanks.

Stephane-D, 2020-03-12:
Definitely yeah, that is the classic approach and going over 16 colors per plane is something you rarely need anyway ;)