Stephane-D/SGDK #154: XGM Loop delay

DenisGrachev, 2020-03-11:
Hi there! There is a delay on a next play after loop vgm with xgm driver. A sample vgm file here:
https://www.dropbox.com/s/mtupgdmkd1180i7/test.zip?dl=0

cheers

Stephane-D, 2020-03-11:
Hi Denis,

Can you tell me where is the loop in the music ? it seems to be located at 40 seconds but i'm not sure and it will help to exhibit the problem. Thanks.

DenisGrachev, 2020-03-11:
Well, yep it loops around a 48-49 sec. There is also defelmask module inside archive to see a loop!

Stephane-D, 2020-03-11:
Ok thanks. Did you used last XGM driver and xgmtool to convert it ?
To be honest it's not really easy to exhibit the issue.. so i'm not sure exactly which instrument is delayed. Is it FM instrument or a PCM one ? Whe using loop the problem is that it may require lot of FM instrument reloading at loop point which may induce some delay. Is it possible to somehow force FM instrument reload in deflemask just before the loop point to see if that help ?

DenisGrachev, 2020-03-11:
Yep, i used latest 1.41 release with a res file like XGM music "sound\music.vgm". I'm not sure which instrument is delayed but a instrument 7 as kick is most noticable. It's a FM instrument. Not sure about command to force instrument reload, will check with an composer!

DenisGrachev, 2020-03-11:
Ok, we made a workaround with arranging a tune. Actually it's not only about loop, there is a loud click even on first start play, looks like too much FM instruments inits. Anyway it can be done by adjusting a tune so i think issue may be closed )

Stephane-D, 2020-03-11:
Oh thanks for sharing the information :)
Just curious, did it was indeed related to FM instrusment loading ? If yes to be honest there is not much which can be done to fix it at runtime as re-loading load of FM instrument can take a while and so delay a bit next note. The only workaround is somehow to dilute instrument loading on several XGM frames to avoid the issue .

DenisGrachev, 2020-03-16:
Yep, looks like a FM instrument loading, we delayed some notes and now it works pretty well. Also sometime XGM_stopPlay makes a tune plays superfast for sometime instead of stop it,i've used xgm_pausePlay to workaround also.

Stephane-D, 2020-03-16:
Ah yeah sometime the stop command doesn't work, i have to investigate that but indeed pause seems to be enough as a workaround.

closed 2020-03-16 by Stephane-D