Stephane-D/SGDK #159: XGM_resumePlay() does not restore note state

evethr, 2020-05-01:
Usually it is not a problem at all, because notes are turned on and off quite frequently in most tracks, getting them restored naturally quite quickly. However, some VGMs are not that cooperative in this aspect: [tension1_4fm.zip](https://github.com/Stephane-D/SGDK/files/4566113/tension1_4fm.zip). This file does not play anything at all after XGM_pausePlay() and XGM_resumePlay() until it fully wraps around.

This issue was noticed in SGDK 1.40, but I believe that it is still present in newer versions.

Stephane-D, 2020-05-03:
In fact this is something we can't really do anything for... This comes from the way about how FM synthesis work. Note can have long decay making them sound for a long time. We you use pause/resumePlay() command, we do not resend lasts key-on commands otherwise it would completely sound wrong and there is no way to restore _decaying notes_ it's why restore() command never restore exactly the music in the exact same state it was.

closed 2020-05-03 by Stephane-D

clbr, 2020-05-04:
Perhaps xgmtool could detect that and do so during the vgm-to-xgm conversion?

Stephane-D, 2020-05-04:
That's definitely not something easy to do (need to look at the different parameter to know if the case can happen), also that would make the music heavier that it should and not all people would like that. I think that is really something that should be done on purpose.

clbr, 2020-05-04:
Please document the limitation in the pause/resume function headers.

Stephane-D, 2020-05-04:
I will do it but again you will *never* see a _resume_ play operation on Sega Mega Drive work as resuming a MP3 music playback. This is something just not possible, you can try to test with Mega Drive game by pausing / unpausing games and you will see that some notes can't be restored until next _key-on_ event..

matteusbeus, 2020-05-04:
You'll note in a large number of games the music just resets and plays from the beginning again.

Dr Matthew Bennion CEng MBCS
NHS England Clinical Entrepreneur,
Digital Mental Health Researcher,
Software Developer & Consultant
E: matthewbennion@hotmail.com
W: http://www.matthewbennion.co.uk

Sent via Outlook for Android<https://aka.ms/ghei36>