Stephane-D/SGDK #170: Problems configuring SGDK on Visual Studio 2005 Professional

Guindows, 2020-08-27:
I was followed the instructions of this video for use SGDK on Visual Studio:
https://www.youtube.com/watch?v=Kx8__lr4Ao4
But, on that video, was used Visual Studio 2017, and I use Visual Studio 2005 Professional (Because my computer is older and I cannot install Visual Studio modern versions on it).
On that video, for use the Sega Mega Drive SGDK codes, it add the SGDK\inc root to "include directory" on "VC++" section on the project properties.
But, on Visual Studio 2005 Professional, I not found that feature that is present on Visual Studio 2017.

Where is that feature on Visual Studio 2005?
Or, How I can configure and use SGDK on Visual Studio 2005?

Stephane-D, 2021-07-08:
Definitely not a bug, just require a proper tutorial :)

thinkOfAnAnswer, 2021-12-10:
> Definitely not a bug, just require a proper tutorial :)

I was afraid you'd say that, haha!
So the game compiles and runs fine, it just stopped compiling straight from VSC, what could've led to that, because I'm very curious!

Also, Stephen. I'm an indie dev and would love to be able to make a future game on the Mega Drive alongside modern systems for my future projects, and using the MEGA CD in some capacity would be a dream, so is there any chance of an expansion to the SGDK library to be compatible with the MEGA CD in the future?

With Terraonion's Mega SD and Krikzz EverDrive, development and testing on actual hardware is now a genuine reality, so I'm curious to know what you're thinking there.

Finally, what is realistically, the largest size a Mega Drive rom file can be? With EverDrives, we're no longer confined to 4MBs, so what would the scientific limit be? I'm asking in case the MEGA CD option is out, we could just take the SNES EverDrive SD card extra capacity approach if that's possible.

Sorry for all the questions, but its been a very exciting week for me working with this thing and a life dream is now a genuine reality thanks to you!

closed 2022-02-08 by Stephane-D