Stephane-D/SGDK #171: Problème importation resources

izzidor, 2020-09-23:
Bonjour,
 j'utilise SGDK avec Visual studio code, je rencontre un problème pour importé une resource "IMAGE".
J'ai créé un fichier .res dans le dossier " res " de mon projet, puis coller à l’intérieur ceci :  IMAGE   bgtile    "tbgtile.png"  0  
A la compilation j'ai l’erreur suivante  :
make : *** [res\resources.rs] error 127

Le fichier resources.h n'est pas généré.

Merci pour votre aide.

izzidor, 2020-09-23:
Extrait du terminal : 
D:\ProgMD\teste>%GDK%\bin\make -f %GDK%\makefile.gen
d:/ProgMD/GenDev/bin/mkdir -p src/boot
d:/ProgMD/GenDev/bin/mkdir -p out
java -jar d:/ProgMD/GenDev/bin/rescomp.jar res/resources.res res/resources.rs -dep out/res/resources.o
d:/ProgMD/GenDev/bin/sh: java: command not found
make: *** [res/resources.rs] Error 127

izzidor, 2020-09-23:
Probléme regler avec l'installation de l'extension " java extension pack "

Stephane-D, 2020-09-24:
Désolé pour la réponse un peu tardive, mais oui SGDK nécessite Java (c'est indiqué dans le fichier readme) pour la compilation des ressources :)

closed 2020-09-24 by Stephane-D

izzidor, 2020-09-24:
Pas de souci, et merci pour ton travail !